analogdevicesinc/ToF #120: Can this software be run in other Cortex A53 platform, like rockchip rk3568

funshine, 2022-07-28:
The Eval platform is i.mx8 m plus. can this sdk/app be run in other ARM platform, like rk3568, rk3588?
Is the software use any gpu/isp/npu specific in i.mx8 m plus? or only eth+mipi+usb3+cpu?
rk3568/rk3588 also have usb3.0 dual role device.
and there are some 4 core A53 platform, without GPU, but have usb3 device(DRD), and mipi, and 5Tops NPU.
Thanks.

andrestraker, 2022-08-05:
Hi, we currently support drivers for the NXP iMX8+ and NVIDIA Jetson. The drivers are V4L2 camera sensor drivers in the context of V4L2 on these platforms. We have not ported the device drivers to the Rockchip devices, but that is doable by a Linux driver engineer with V4L2 experience. At this point, the SDK is more or less generic user space Linux. The core difficulty is the depth compute library. This would need to be built and tested on the Rockchip device. Note, we are working on an open source solution for the ATDF3175D (where the depth ISP, ADSD3500, is in the signal chain). With this open source version porting to the Rockchip will be an easier task for customers such as yourself.

funshine, 2022-08-09:
Thank you! sounds great!
so, adsd3500 is an ISP,  and ATDF3175D = (laser part) + adsd3100 --> adsd3500(isp) --> mipi?
The depth compute lib ( tofi_compute, tofi_config ) and ccb lib are embeded in ADSD3500 ISP chip right?
if I buy the atdf3175d module, only i2c driver part needs to be ported to other platform, like a rgb sensor?

closed 2024-04-26 by andrestraker