analogdevicesinc/ToF #127: ROS2 Wrapper for EVAL-ADTF3175D-NXZ

gsingh81, 2022-08-12:
we are planning to use EVAL-ADTF3175D-NXZ on ROS2 Humble. However it seems current SDK ROS wrapper is based on ROS1 https://github.com/analogdevicesinc/ToF/tree/master/bindings/ros
a) Is EVAL-ADTF3175D-NXZ tested with ROS1 ?
b) please let us know if there is any plan or roadmap to provide ROS2 wrapper ?
c) If not, do you recommend to use ROS1/ROS2 bridge ? will the interfaces supported by https://github.com/ros2/ros1_bridge will work with ROS1 wrapper ?

Thanks
gurpreet

andrestraker, 2022-08-24:
In the coming weeks we will publish a schedule for ROS2 support, stay tuned.

dNechita, 2023-06-30:
Hi!
The ROS2 support can be found at: https://github.com/analogdevicesinc/tof-ros2

closed 2023-06-30 by dNechita