analogdevicesinc/ToF #131: Linux build support for EVAL-ADTF3175D-NXZ

meetgandhi21, 2022-08-23:
We have EVAL-ADTF3175D-NXZ kit and trying to build it for Linux PC.

We are getting below error while building ToF software suite for EVAL-ADTF3175D-NXZ as per the instruction provided [here](https://github.com/analogdevicesinc/ToF/blob/v3.0.0/doc/itof/linux_build_instructions.md) 
 

make[2]: *** No rule to make target '../../libs/libtofi_compute.so', needed by 'sdk/libaditof.so.3.0.0'.  Stop.

CMakeFiles/Makefile2:153: recipe for target 'sdk/CMakeFiles/aditof.dir/all' failed
make[1]: *** [sdk/CMakeFiles/aditof.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2

Can you please provide some reference using which we can get libtofi_compute.so library source ?
Is there any other .so file for which we need library source to build SDK ?

Full logs:

Build log with error:
meet@AHMLPT1819:~/project/robo_23/ToF/build$ cmake -DUSE_ITOF=1 -DWITH_EXAMPLES=off ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.a;-lpthread (found suitable version "3.9.0", minimum required is "3.9.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/meet/project/robo_23/ToF/build
meet@AHMLPT1819:~/project/robo_23/ToF/build$ make
Scanning dependencies of target ccb
[  3%] Building C object sdk/common/adi/ccb/CMakeFiles/ccb.dir/src/ccb.c.o
[  6%] Linking C static library libccb.a
[  6%] Built target ccb
[ 10%] Running cpp protocol buffer compiler on src/connections/usb/usb_buffer.proto
Scanning dependencies of target aditof
[ 13%] Building CXX object sdk/CMakeFiles/aditof.dir/src/filters_factory.cpp.o
[ 17%] Building CXX object sdk/CMakeFiles/aditof.dir/src/frame.cpp.o
[ 20%] Building CXX object sdk/CMakeFiles/aditof.dir/src/frame_operations.cpp.o
[ 24%] Building CXX object sdk/CMakeFiles/aditof.dir/src/sensor_enumerator_factory.cpp.o
[ 27%] Building CXX object sdk/CMakeFiles/aditof.dir/src/system.cpp.o
[ 31%] Building CXX object sdk/CMakeFiles/aditof.dir/src/system_impl.cpp.o
[ 34%] Building CXX object sdk/CMakeFiles/aditof.dir/src/utils.cpp.o
[ 37%] Building CXX object sdk/CMakeFiles/aditof.dir/src/cameras/itof-camera/adsd3100_configuration.cpp.o
In file included from /home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/adsd3100_configuration.h:28:0,
                 from /home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/adsd3100_configuration.cpp:25:
/home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/adsd3100_configuration.cpp: In member function â€˜aditof::Status aditof::ADSD3100Configuration::parseConfigFile(const char*)â€™:
/home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/aditof_common.h:28:75: warning: value computed is not used [-Wunused-value]
 #define fopen_s(pFile,filename,mode) ((*(pFile))=fopen((filename),(mode)))==NULL
                                                                           ^
/home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/adsd3100_configuration.cpp:65:5: note: in expansion of macro â€˜fopen_sâ€™
     fopen_s(&fid, file_name, "rb");
     ^~~~~~~
[ 41%] Building CXX object sdk/CMakeFiles/aditof.dir/src/cameras/itof-camera/calibration_itof.cpp.o
In file included from /home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/adsd3100_configuration.h:28:0,
                 from /home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/calibration_itof.h:27,
                 from /home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/calibration_itof.cpp:24:
/home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/calibration_itof.cpp: In member function â€˜aditof::Status CalibrationItof::writeConfiguration(const string&)â€™:
/home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/aditof_common.h:28:75: warning: value computed is not used [-Wunused-value]
 #define fopen_s(pFile,filename,mode) ((*(pFile))=fopen((filename),(mode)))==NULL
                                                                           ^
/home/meet/project/robo_23/ToF/sdk/src/cameras/itof-camera/calibration_itof.cpp:136:5: note: in expansion of macro â€˜fopen_sâ€™
     fopen_s(&fd, (const char *)configurationFile.c_str(), "rb");
[... 167 lines cut ...]
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

meet@AHMLPT1819:~$ cmake --version
cmake version 3.18.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
meet@AHMLPT1819:~$ g++ --version
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

meet@AHMLPT1819:~$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

andrestraker, 2022-08-23:
We are waiting for a license update to support depth compute library on Linux. For now, you can build the SDK USE_DEPTH_COMPUTE_STUBS  to ON in CMakeLists.txt. When the license update has been completed we will make an announcement.

https://github.com/analogdevicesinc/ToF/blob/master/CMakeLists.txt#L28

meetgandhi21, 2022-08-24:
Thanks for the prompt response.
I was able to build the SDK using the provided option.
Just wanted to check,
As we're using some stubs will it have any impact on data published for ROS bindings mentioned [here](https://github.com/analogdevicesinc/ToF/tree/master/bindings/ros#published-topics)

Ruyou2529, 2022-09-06:
When I compile data collect in example, the following error occurs:
/usr/local/include/aditof/version.h:46:77: error: ‘ADITOFSDK_GIT_BRANCH’ was not declared in this scope
 ic inline const std::string getBranchVersion() { return std::string(ADITOFSDK_GIT_BRANCH); }
/usr/local/include/aditof/version.h: In function ‘const string aditof::getCommitVersion()’:
/usr/local/include/aditof/version.h:47:77: error: ‘ADITOFSDK_GIT_COMMIT’ was not declared in this scope
 ic inline const std::string getCommitVersion() { return std::string(ADITOFSDK_GIT_COMMIT); }

Can you please provide some advice?

SeptimiuVana, 2022-09-07:
Hello @Ruyou2529 

We have fixed this error, please update the sdk to the latest version.

HAKUTA-design, 2022-09-14:
Hello @SeptimiuVana 
When I compile the ToF-3.2.0 sdk examples and run it, I encountered the following problems:

My CMake command is :
`
.../ToF-master/build$ cmake \
-DWITH_EXAMPLES=on \
-DWITH_NETWORK=on \
-DWITH_OFFLINE=on \
-DUSE_DEPTH_COMPUTE_STUBS=on \
-DCMAKE_PREFIX_PATH="/opt/glog;/opt/protobuf;/opt/websockets" ..
`
it can compile successfully:
`
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
Downloading raw frames into /home/xita/Packages/ToF-master/build/resources/offline
-- Found Protobuf: /opt/protobuf/lib/libprotobuf.a;-lpthread (found suitable version "3.9.0", minimum required is "3.9.0") 
-- Found OpenCV: /usr (found version "4.2.0") 
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  examples/tof-viewer/CMakeLists.txt:47 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xita/Packages/ToF-master/build
`

1. When I tried to run the data_collect, it reported the following error: 
**Could not find MX25U6435F while looking for storage**
`
I0914 14:43:55.091557 44832 main.cpp:257] SDK version: 3.2.0 | branch:  | commit: 
I0914 14:43:55.112407 44832 main.cpp:393] Output folder: ./data_output
I0914 14:43:55.112437 44832 main.cpp:394] Mode: 10
I0914 14:43:55.112447 44832 main.cpp:395] Number of frames: 4
I0914 14:43:55.112455 44832 main.cpp:396] Json file: ./config/config_default.json
I0914 14:43:55.112462 44832 main.cpp:397] Frame type is: raw
I0914 14:43:55.112470 44832 main.cpp:398] Warm Up Time is: 5 seconds
I0914 14:43:55.112506 44832 system_impl.cpp:86] SDK built with websockets version:3.1.0
I0914 14:43:55.112520 44832 system_impl.cpp:95] Creating offline sensor.
W0914 14:43:55.113065 44832 camera_itof.cpp:110] Could not find MX25U6435F while looking for storage
I0914 14:43:55.113404 44832 camera_itof.cpp:137] Initializing camera
I0914 14:43:55.113574 44832 camera_itof.cpp:1491] Current sensor firmware is: ./config/camera_configuration_walden.cfg
I0914 14:43:55.113591 44832 camera_itof.cpp:1508] Current calibration file is: ./config/camera_calibration.ccb
I0914 14:43:55.113610 44832 camera_itof.cpp:1564] Current Depth ini file is: ./config/RawToDepth.ini
W0914 14:43:55.113734 44832 camera_itof.cpp:1494] Duplicate firmware file ignored: ./config/camera_configuration_walden.cfg
W0914 14:43:55.113782 44832 camera_itof.cpp:1511] Duplicate calibration file ignored: ./config/camera_calibration.ccb
I0914 14:43:55.127542 44832 camera_itof.cpp:269] Camera initialized
I0914 14:43:55.127615 44832 main.cpp:444] SD card image version: 
I0914 14:43:55.127624 44832 main.cpp:445] Kernel version: 
I0914 14:43:55.127629 44832 main.cpp:446] U-Boot version: 
I0914 14:43:55.127737 44832 camera_itof.cpp:345] Chosen mode: mp
I0914 14:43:55.127877 44832 main.cpp:573] FSF File name: ./data_output/raw_frames_20220914144355.fsf
I0914 14:43:55.127897 44832 camera_itof.cpp:296] USEQ running. Skip CFG & CCB programming step
E0914 14:43:55.128115 44832 frame_impl.cpp:132] raw is not supported by this frame!
Segmentation fault (core dumped)
`
My command is:
`
$ cd ./ToF-master/build/examples/data_collect
$ ./data_collect --f "./data_output" --m 10 --n 4 "./config/config_default.json" --fsf 1 --wt 5
`

2. When I run tof-viewer, **Camera:Log** window also reported the following error: 
**Could not find MX25U6435F while looking for storage**
My command is:
`
$ cd ./ToF-master/build/examples/tof-viewer
$ ./ADIToFGUI
`

3. When I added **aditof-demo** example to CMakeLists and tried to compile and make it, the following error occurs:
`
xita@xita:~/ToF-master/build$ make
[  2%] Built target ccb
[  4%] Built target tofi_config
[  7%] Built target tofi_compute
[ 49%] Built target aditof
[ 50%] Linking CXX executable aditof-demo
Error copying directory from "/home/xita/Packages/ToF-master/../libs" to "/home/xita/Packages/ToF-master/build/examples/aditof-demo/.".
make[2]: *** [examples/aditof-demo/CMakeFiles/aditof-demo.dir/build.make:181: examples/aditof-demo/aditof-demo] Error 1
make[2]: *** Deleting file 'examples/aditof-demo/aditof-demo'
make[1]: *** [CMakeFiles/Makefile2:476: examples/aditof-demo/CMakeFiles/aditof-demo.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
`
I've changed the "ToF-master/examples/CMakeLists.txt" to:
`
cmake_minimum_required(VERSION 3.0)
project(examples)

add_subdirectory(aditof-demo)
add_subdirectory(first-frame)
add_subdirectory(data_collect)
add_subdirectory(imshow-jetson)
add_subdirectory(imshow-xavier)

if (NOT NXP)
    add_subdirectory(tof-viewer)
endif()
...
`

Can you please provide me some advice?

SeptimiuVana, 2022-09-14:
Hello,

Please set the -DWITH_OFFLINE cmake flag to 0 and rebuild the sdk. And for this eval kit please use the config_walden_3500_nxp.json file from the config folder instead of the config_default.json one.

This will help you extract raw frames from the camera using data-collect

HAKUTA-design, 2022-09-14:
> Hello,
> 
> Please set the -DWITH_OFFLINE cmake flag to 0 and rebuild the sdk. And for this eval kit please use the config_walden_3500_nxp.json file from the config folder instead of the config_default.json one.
> 
> This will help you extract raw frames from the camera using data-collect

When I tried your method, a new error occurs: **No cameras found**

```
I0914 17:00:26.195742 57420 main.cpp:257] SDK version: 3.2.0 | branch:  | commit: 
I0914 17:00:26.200780 57420 main.cpp:393] Output folder: ./data_output
I0914 17:00:26.200789 57420 main.cpp:394] Mode: 10
I0914 17:00:26.200809 57420 main.cpp:395] Number of frames: 4
I0914 17:00:26.200814 57420 main.cpp:396] Json file: ./config/config_walden_3500_nxp.json
I0914 17:00:26.200832 57420 main.cpp:397] Frame type is: raw
I0914 17:00:26.200835 57420 main.cpp:398] Warm Up Time is: 5 seconds
I0914 17:00:26.200848 57420 system_impl.cpp:86] SDK built with websockets version:3.1.0
I0914 17:00:26.200894 57420 usb_sensor_enumerator.cpp:56] Looking for USB connected sensors
W0914 17:00:26.201048 57420 main.cpp:422] No cameras found
```

My command is :

```
xita@xita:~/Packages/ToF-master/build/examples/data_collect$ ./data_collect --f "./data_output" --m 10 --n 4 "./config/config_walden_3500_nxp.json" --fsf 1 --wt 5
```