analogdevicesinc/ToF #134: Unable to get EVAL-ADTF3175D-NXZ data for using ADIToFGUI Tool

meetgandhi21, 2022-08-26:
Hi,

We are having EVAL-ADTF3175D-NXZ eval system and are trying to get data using the ADIToFGUI Windows Tool as per instruction provided [here](https://wiki.analog.com/resources/eval/user-guides/eval-adsd3100-nxz-gui). 
However, we are not able to get any information from the Camera.
We also tried the Troubleshooting steps provided [here](https://wiki.analog.com/resources/eval/user-guides/aditofgui_ts) but they didn't work.
Following are some of the logs we found in GUI App:

```
I0826 18:25:42.834313  1528 ADIToF.cpp:83] SDK version: 3.0.0 | branch: adsd3500 | commit: 5eef987
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0826 18:25:46.657864  1528 system_impl.cpp:80] SDK built with websockets version:3.1.0
I0826 18:25:46.658862  1528 usb_sensor_enumerator.cpp:55] Looking for USB connected sensors
I0826 18:25:46.685214  1528 system_impl.cpp:123] SDK built with websockets version:3.1.0
I0826 18:25:46.686213  1528 network_sensor_enumerator.cpp:53] Looking for sensors over network
[2022/08/26 18:25:46:6900] NOTICE: Creating Vhost 'default' (serving disabled), 1 protocols, IPv6 off
I0826 18:25:51.933068 17564 ADIMainWindow.cpp:1128] Preparing camera. Please wait...
I0826 18:25:51.934067 17564 camera_itof.cpp:133] Initializing camera
[2022/08/26 18:25:51:9456] NOTICE: Creating Vhost 'default' (serving disabled), 1 protocols, IPv6 off
I0826 18:25:57.094662 17564 camera_itof.cpp:206] Found Depth ini file: ../config/RawToDepthAdsd3500_mp.ini
I0826 18:25:57.094662 17564 camera_itof.cpp:206] Found Depth ini file: ../config/RawToDepthAdsd3500_qmp.ini
I0826 18:25:57.094662 17564 camera_itof.cpp:218] Current Depth ini file is: ../config/RawToDepthAdsd3500_mp.ini
E0826 18:25:57.119984 17564 camera_itof.cpp:261] Failed to read intrinsics for adsd3500!
E0826 18:25:57.119984 17564 ADIMainWindow.cpp:1152] Could not initialize camera!
```
Can anyone help to troubleshoot the same?

SeptimiuVana, 2022-08-30:
Hello, 

We are looking into this and we will come back with an answer as soon as possible.

meetgandhi168, 2023-06-02:
With the latest SDK release, I was able to capture and visualize the Frame on Linux platform.
Closing the issue.

closed 2023-06-02 by meetgandhi168