analogdevicesinc/ToF #144: Pulsatrix eval board I2C communication issues

TK-cognex, 2022-09-15:
Hi,

I hope this is the right forum for this topic.
We have started the bring-up of the ADSD3500 Interposer board (EVAL ASDS35000-1, Rev C) and encounter some issues in the communication to the Pulsatrix IC.
The setup is our own development kit with NXP imx8mp processor and all lines connected tho the Pulsatrix board via board-to-board connector. 

Current state is:
- Power rails checked:
  - TP1: VCC_5V = 5V
  - TP2: Vaux = 22V
  - TP5: VSYS = 5V
  
We want to use Master Mode 1, hence we tested bootstrapping:
- BS0: GND by hogging i2c expander pin OC0 to low - Master boot mode
- BS1: GND by hogging i2c expander pin OC1 to low (with BS3) 00: I2C slave interface static address 0x70/0x71
- BS2: GND by hogging i2c expander pin OC2 to low - 24MHz as used 
- BS3: GND by hogging i2c expander pin OC3 to low - see BS1
- BS4: GND by hogging i2c expander pin OC4 to low - SPI don't care, we use I2C 
- BS5: GND by hogging i2c expander pin OC5 to low - boot readiness via host_irq 

Side note: Documentation shows different pins for host_irq/boot readniess in different master modes, but in the schematic of the integration guide p.46 GPIO15 is host_irq. GPIO_19 like in master mode 2 would conflict with bootstrapping.

In this mode we see the Blue LED being constantly on and the Pulsatrix does not appear on the I2C bus. We are tracing the signals and the address is not acked. 
GPIO15: boot readniess is constantly high
Question: Is the blue LED supposed to be on during boot or at all times? In other words what is the state of Fsync?

We tried to remap the Pulsatrix to address 0x80/0x81 by changing the bootstapping, such that:
- BS3: 1.8V by hogging i2c expander pin OC3 to high 

With this bootstrapping we see that

- After reset: 
  - the Blue LED is on for some time, but the device is not answering on I2C address 
  - After some seconds the LED goes off (boot indicator?) and the device appears at I2C address 0x40(0x80)
- Communication via i2ctransfer cmd (i2ctransfer 2 w2@0x40 0x01 0x12 r2) shows    
  - in approx. 20% of the transfers results of either 0x1 or 0x0, but traced I2C communication looks ok 
  - 80% of the I2C communnications are failing as the Pulsatrix is sending only one byte and the second byte is missed (address and reg address looks ok)
  
Any idea what we are missing?

Thanks 
Torsten

tmanthey, 2023-04-27:
Hi Torsten, 
did you find a solution for this?
If you read byte values the order is 6x failing followed by 0x01 0x00 then start to fail again.
If you run i2cdetect the device is visible only after returning 0x01 0x00 for 2 consecutive queries.

andrestraker, 2024-04-26:
User did not reply. Closing.

closed 2024-04-26 by andrestraker