analogdevicesinc/ToF #146: Linux build support for ToF-3.2.0

HAKUTA-design, 2022-09-22:
When I compile the ToF-3.2.0 sdk examples and run it, I encountered some problems:

My CMake command is : (I have already installed python3.8, opencv4.2 and ros2)

```shell
cd ToF-master
mkdir build & cd build
cmake -DWITH_EXAMPLES=on -DUSE_DEPTH_COMPUTE_STUBS=on -DCMAKE_PREFIX_PATH="/opt/glog;/opt/protobuf;/opt/websockets" ..
make
```

it can compile and make successfully.

and then cd to examples:

```shell
cd examples/tof-viewer/
./ADIToFGUI
```

1. when tried to run `./ADIToFGUI` , it reported the following error:

   ```
   Network interface is not enabled. Please rebuild the SDK with the option WITH_NETWORK=on
   ```

   So, I follow the tips and rebuild it, my cmake command is:

   ```shell
   cmake -DWITH_EXAMPLES=on -DWITH_NETWORK=on -DUSE_DEPTH_COMPUTE_STUBS=on -DCMAKE_PREFIX_PATH="/opt/glog;/opt/protobuf;/opt/websockets" ..
   make
   ```

   It compiles successfully, I try to run it again:

   ```shell
   cd examples/tof-viewer/
   ./ADIToFGUI
   ```

   **When I clicked `Open Device` then clicked `Play`, it crash !!!**

   The terminal display:

   ```shell
   Conn established
   Connection Closed
   Conn established
   Segmentation fault (core dumped)
   ```

2. When I tried to run the `data_collect` example, it reported the following error:
    **Could not find MX25U6435F while looking for storage !**

   My command is :

   ```shell
   cd examples/data_collect
   ./data_collect --f "./data_output" --m 10 --n 4 "./config/config_walden_3500_nxp.json" --fsf 1 --wt 5
   ```

   The terminal display:

   ```shell
   I0922 09:37:16.754204 18863 main.cpp:257] SDK version: 3.2.0 | branch:  | commit: 
   I0922 09:37:16.771416 18863 main.cpp:393] Output folder: ./data_output
   I0922 09:37:16.771428 18863 main.cpp:394] Mode: 10
   I0922 09:37:16.771432 18863 main.cpp:395] Number of frames: 4
   I0922 09:37:16.771435 18863 main.cpp:396] Json file: ./config/config_walden_3500_nxp.json
   I0922 09:37:16.771438 18863 main.cpp:397] Frame type is: raw
   I0922 09:37:16.771441 18863 main.cpp:398] Warm Up Time is: 5 seconds
   I0922 09:37:16.771456 18863 system_impl.cpp:86] SDK built with websockets version:3.1.0
   I0922 09:37:16.771483 18863 usb_sensor_enumerator.cpp:56] Looking for USB connected sensors
   W0922 09:37:16.771628 18863 main.cpp:422] No cameras found
   ```

   

Can you please provide me some advice?

Ruyou2529, 2022-09-23:
Can this camera only be connected to the real system? Why can't I find the device number after connecting the camera to the virtual machine?

rbudai98, 2022-09-23:
HI! We have managed to reproduce the error. Could you please let us know if you were building the SDK on the NXP or on host machine?
In case of the host machine the network communication is working if you specify the IP address of the camera, such as: 
`./data_collect --ip "10.42.0.1" --f "./data_output" --m 10 --n 4 "./config/config_walden_3500_nxp.json" --fsf 1 --wt 5
`

rbudai98, 2022-09-23:
Also be aware that in this case network means ethernet via usb, so the type C connection is enough for the previously mentioned example

HAKUTA-design, 2022-09-26:
We are building it on host machine. When I tried your method to specify the IP address:
`./data_collect --ip "10.42.0.1" --f "./data_output" --m 10 --n 4 "./config/config_walden_3500_nxp.json" --fsf 1 --wt 5`
it made another mistake: 
**W0926 10:04:36.494877 16799 camera_itof.cpp:1494] Duplicate firmware file ignored: ./config/NXP_MIPI4kSuperframe.cfg**
**W0926 10:04:36.494958 16799 camera_itof.cpp:1511] Duplicate calibration file ignored: ./config/camera_calibration_adsd3500.ccb**
**I0926 10:04:36.499728 16799 camera_itof.cpp:262] Failed to set fps!**
**Could not initialize camera!**
Complete information display of terminal:
```
I0926 10:04:36.400498 16799 main.cpp:257] SDK version: 3.2.0 | branch:  | commit: 
I0926 10:04:36.405541 16799 main.cpp:393] Output folder: ./data_output
I0926 10:04:36.405551 16799 main.cpp:394] Mode: 10
I0926 10:04:36.405555 16799 main.cpp:395] Number of frames: 4
I0926 10:04:36.405558 16799 main.cpp:396] Json file: ./config/config_walden_3500_nxp.json
I0926 10:04:36.405561 16799 main.cpp:397] Frame type is: raw
I0926 10:04:36.405565 16799 main.cpp:398] Warm Up Time is: 5 seconds
I0926 10:04:36.405568 16799 main.cpp:401] Ip address is: 10.42.0.1
I0926 10:04:36.405580 16799 system_impl.cpp:130] SDK built with websockets version:3.1.0
I0926 10:04:36.405612 16799 network_sensor_enumerator.cpp:53] Looking for sensors over network
[2022/09/26 10:04:36:4057] NOTICE: Creating Vhost 'default' (serving disabled), 1 protocols, IPv6 off
Conn established
Connection Closed
I0926 10:04:36.425805 16799 camera_itof.cpp:137] Initializing camera
[2022/09/26 10:04:36:4258] NOTICE: Creating Vhost 'default' (serving disabled), 1 protocols, IPv6 off
Conn established
I0926 10:04:36.494575 16799 camera_itof.cpp:203] Current adsd3500 firmware version is: 3.1.0.0
I0926 10:04:36.494589 16799 camera_itof.cpp:206] Current adsd3500 firmware git hash is: c091709d4a890743a92e8aff3b80dc3f8859aa7c
I0926 10:04:36.494683 16799 camera_itof.cpp:1491] Current sensor firmware is: ./config/NXP_MIPI4kSuperframe.cfg
I0926 10:04:36.494690 16799 camera_itof.cpp:1508] Current calibration file is: ./config/camera_calibration_adsd3500.ccb
I0926 10:04:36.494704 16799 camera_itof.cpp:1553] Found Depth ini file: ./config/RawToDepthAdsd3500_qmp.ini
I0926 10:04:36.494742 16799 camera_itof.cpp:1553] Found Depth ini file: ./config/RawToDepthAdsd3500_mp.ini
I0926 10:04:36.494765 16799 camera_itof.cpp:1564] Current Depth ini file is: ./config/RawToDepthAdsd3500_mp.ini
W0926 10:04:36.494877 16799 camera_itof.cpp:1494] Duplicate firmware file ignored: ./config/NXP_MIPI4kSuperframe.cfg
W0926 10:04:36.494958 16799 camera_itof.cpp:1511] Duplicate calibration file ignored: ./config/camera_calibration_adsd3500.ccb
I0926 10:04:36.499728 16799 camera_itof.cpp:262] Failed to set fps!
E0926 10:04:36.499739 16799 main.cpp:437] Could not initialize camera!
Connection Closed
```
My USB connection is proper, and the config file passed in also correct.
I don't know how to fix it, can you please provide me some advice?

rbudai98, 2022-09-26:
Yes, the warning can be ignored

closed 2022-09-26 by rbudai98