analogdevicesinc/ToF #163: Frames are not shown in ADI ToF GUI App

meetgandhi21, 2022-10-10:
We are running the ToF SDK with Depth compute stub enabled for x86_64 Ubuntu machine as per instructions given [here](https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__https%3A%2Fnam04.safelinks.protection.outlook.com%2F%3Furl%3Dhttps*3A*2F*2Fgithub.com*2Fanalogdevicesinc*2FToF*2Fissues*2F131%26data%3D05*7C01*7CAmit.Gupta*40einfochips.com*7Ce66f762bd89b420a2f3708daa5faaee3*7C0beb0c359cbb4feb99e5589e415c7944*7C1*7C0*7C638004793347183634*7CUnknown*7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0*3D*7C3000*7C*7C*7C%26sdata%3DmWyXISJGxWAvi6fr5JAosfaVWvtM9ulwizyjRn0aBRQ*3D%26reserved%3D0__%3BJSUlJSUlJSUlJSUlJSUlJSUlJSUlJSU!!A3Ni8CS0y2Y!9qV1B2RO5wtUDTWqTTtSDk_Zlcput6LbFOKqAEBFtVxwDONT7nqdxnO8Ukzwu10p44IgZS9kBb_TdiQ_5gfzYQKgdA%24&data=05%7C01%7CMeet.Gandhi%40einfochips.com%7C3e79e1a4b1654e8402b608daa6152830%7C0beb0c359cbb4feb99e5589e415c7944%7C1%7C0%7C638004907045583593%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=gRycNaRX8uLZJcHIbjQAFr7ok%2BL5Fvzm%2BZvCHn9MOeg%3D&reserved=0)
We tried to run the ADIToFGUI example application to capture the frames. But it gives segmentation fault in both mp and qmp mode. 
We took a walk through the code and found that the crash happens as there is no implementation for the [Depth-comupte ](https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__https%3A%2Fnam04.safelinks.protection.outlook.com%2F%3Furl%3Dhttps*3A*2F*2Fgithub.com*2Fanalogdevicesinc*2FToF*2Ftree*2Fmaster*2Fsdk*2Fcommon*2Fadi*2Fdepth-compute-stub*2Fsrc%26data%3D05*7C01*7CAmit.Gupta*40einfochips.com*7Ce66f762bd89b420a2f3708daa5faaee3*7C0beb0c359cbb4feb99e5589e415c7944*7C1*7C0*7C638004793347183634*7CUnknown*7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0*3D*7C3000*7C*7C*7C%26sdata%3D5KFkbJ*2FpAwV3w*2BfmVdZzUYhGlcDr*2Bkv1Ip692CCoc14*3D%26reserved%3D0__%3BJSUlJSUlJSUlJSUlJSUlJSUlJSUlJSUlJSUlJSUlJQ!!A3Ni8CS0y2Y!9qV1B2RO5wtUDTWqTTtSDk_Zlcput6LbFOKqAEBFtVxwDONT7nqdxnO8Ukzwu10p44IgZS9kBb_TdiQ_5gdLdYXFxw%24&data=05%7C01%7CMeet.Gandhi%40einfochips.com%7C3e79e1a4b1654e8402b608daa6152830%7C0beb0c359cbb4feb99e5589e415c7944%7C1%7C0%7C638004907045583593%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=2xTLakOQI8zB1ECX6zvzz6v%2BRsQ5cOnMI5rPhgDUCcA%3D&reserved=0)stub.
- Is there anyway using which we can disable the same until the license update gets completed? (Refer to this [github discussion](https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__https%3A%2Fnam04.safelinks.protection.outlook.com%2F%3Furl%3Dhttps*3A*2F*2Fgithub.com*2Fanalogdevicesinc*2FToF*2Fissues*2F131%26data%3D05*7C01*7CAmit.Gupta*40einfochips.com*7Ce66f762bd89b420a2f3708daa5faaee3*7C0beb0c359cbb4feb99e5589e415c7944*7C1*7C0*7C638004793347183634*7CUnknown*7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0*3D*7C3000*7C*7C*7C%26sdata%3DmWyXISJGxWAvi6fr5JAosfaVWvtM9ulwizyjRn0aBRQ*3D%26reserved%3D0__%3BJSUlJSUlJSUlJSUlJSUlJSUlJSUlJSU!!A3Ni8CS0y2Y!9qV1B2RO5wtUDTWqTTtSDk_Zlcput6LbFOKqAEBFtVxwDONT7nqdxnO8Ukzwu10p44IgZS9kBb_TdiQ_5gfzYQKgdA%24&data=05%7C01%7CMeet.Gandhi%40einfochips.com%7C3e79e1a4b1654e8402b608daa6152830%7C0beb0c359cbb4feb99e5589e415c7944%7C1%7C0%7C638004907045583593%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=gRycNaRX8uLZJcHIbjQAFr7ok%2BL5Fvzm%2BZvCHn9MOeg%3D&reserved=0) for more details)
- For depth compute libraries, do we need specific license for Linux ? from whom we can get these libraries ?

andrestraker, 2022-10-26:
Linux support will be offically available by early next week (week of 10/31) at the latest.

andrestraker, 2022-11-01:
The release has been posted, see: https://github.com/analogdevicesinc/ToF/releases/tag/v3.2.0

meetgandhi168, 2023-06-02:
Was able to visualize the Frame with latest SDK on Linux platform.
Closing the issue.

closed 2023-06-02 by meetgandhi168