analogdevicesinc/ToF #180: can not able to capture image via USB  - eval-adtf3175-nxz

VishalRaval-D, 2022-10-21:
Hello, I am facing some problems in running tof-viewer example as well ros2 code in Linux.

I am trying to capture images via USB but could not found camera. 
I tried many times USB adapter  connect and reconnect process but could not get success.
Can you please help me here is there any process or steps missed  for USB connection.
Please find logs::
I1021 17:13:07.937594 14346 aditof_utils.cpp:75] No ip provided, attempting to connect to the camera through USB
I1021 17:13:07.937995 14346 aditof_utils.cpp:105] Started camera intialization
I1021 17:13:07.938016 14346 system_impl.cpp:86] SDK built with websockets version:3.2.3
I1021 17:13:07.938186 14346 usb_sensor_enumerator.cpp:56] Looking for USB connected sensors
 W1021 17:13:07.938486 14346 **aditof_utils.cpp:117] No cameras found**

**- Note-I am able to capture images via network**

andrestraker, 2022-10-26:
At this point the camera is only available as a network device. The UVC interface will be introduced in the next month, pending no complications.

VishalRaval-D, 2022-11-02:
Thanks for your revert. We eagerly waiting for UVC Interface feature.

meetgandhi-eic, 2023-05-05:
Hi,

Is this feature available in the latest release v4.1.1?

dNechita, 2023-05-22:
Hi. No but we are close to release 4.2.0 where UVC will be available (Linux only). The 4.2.0 has been already published as a pre-release.

meetgandhi-eic, 2023-05-23:
Hi @dNechita 
Thanks for the update.
We will check at our end and get back to you.

closed 2024-04-26 by andrestraker