analogdevicesinc/ToF #204: Build failed on windows - ssize_t

daniyalyasin93, 2022-11-08:
The build fails on windows for target aditofpython with Microsoft Visual Studio 2019 win64.

The error is:

```
aditofpython.cpp(142,25): error C2065: 'ssize_t': undeclared identifier [D:\Work\Repos\ToF\scripts\windows\build\bindings\python\aditofpython.vcxproj]
proj]
```

The solution is to include the header BaseTsd.h. This is described here: https://github.com/ycm-core/YouCompleteMe/issues/3949#issuecomment-944786867

dNechita, 2022-11-08:
Hi! Can you please share the Python version that you are using?

daniyalyasin93, 2022-11-11:
Hi
I have python 3.10.1

dNechita, 2022-11-14:
I use python 3.9. I didn't update to anything higher due to this error.
Thank you for the fix!

closed 2022-11-14 by dNechita

daniyalyasin93, 2022-11-14:
Thank you for approving. Glad to be able to contribute