analogdevicesinc/ToF #246: libtofi_compute.so missing for arm64 (Nvidia Jetson)

D3-aavery, 2023-01-19:
@andrestraker mentioned suport for Nvidia Jetson in [https://github.com/analogdevicesinc/ToF/issues/120](url), but I can't find a way to obtain libtofi_compute.so for arm64.

If this is not available, what is the impact of using USE_DEPTH_COMPUTE_STUBS or USE_DEPTH_COMPUTE_OPENSOURCE?

D3-aavery, 2023-01-19:
Hi @andrestraker, I'm assuming you're referring to the following line in those instructions:
cp ~/Analog\ Devices/TOF_Evaluation_ADTF3175D-Rel3.2.0/bin/*tofi* .

However, this depends on successfully running the installer first, right? The installer (adi-tof-adtf3175d-rel3.2.0) is only provided for x86 (aka AMD64). How do I obtain TOF_Evaluation_ADTF3175D-Rel3.2.0, and thus libtofi_compute.so, for arm64?

andrestraker, 2023-01-19:
I will make clear in the readme.md where to get the ARM64 versions.

We include the installers for these in the image file - see the 'image' folder.

Once downloaded, navigate to NXP-Img-Rel3.2.0-ADTF3175D-FB2C93AE\NXP-Img-Rel3.2.0-ADTF3175D\depth_compute.

See the README.md to understand the difference between the packages.

Please note, these installers are Windows based, so you will need to install there first, then bring the .so files back to Linux.

These installers are required since you need to accept the SLA for these versions of the depth compute libraries before being able to use them.

D3-aavery, 2023-01-19:
@andrestraker I downloaded and ran the Windows installer. This allows me to then download TOF_DepthComputeEngine_Linux-Rel3.4.1, but there is only an x86 version of libtofi_compute.so in 'lib' and 'prebuilt_binaries'.

pjdowner, 2023-02-22:
I'm not running on an nvidia jetson, I am on arm64, so maybe this will help, but I am using a script which is apparently intended for an NXP imx8, therefore you may need to do more investigation...

I ran into the same issue, following the documented build instructions I couldn't re-build the SDK on my target device as I was missing libtofi_compute.so. However I knew that the SDK is built when the OS image is built. After digging through the build scripts I found that there is a script which will build everything, I haven't investigated how it builds or gets libtofi but on your target device try the following...

cd ToF/scripts/nxp
./setup.sh

The output should be in ./build

andrestraker, 2023-02-22:
@pjdowner can you email **tof@analog.com** with this request. The AMD64 will be then made available to you.

andrestraker, 2023-06-09:
Hi,

The libraries are available via the evaluation software. If you install the latest eval kit release (4.2.0 - https://github.com/analogdevicesinc/ToF/releases/tag/v4.2.0) you can get the libraries via the highlighted installer shown in the image below.

image

closed 2023-06-30 by dNechita