analogdevicesinc/ToF #340: execution gets stuck after calling TofiCompute

meetgandhi-eic, 2023-06-02:
Hi,

I'm using SDK [Release v4.1.1](https://github.com/analogdevicesinc/ToF/releases/tag/v4.1.1). We have created a test application based on [first-frame example application](https://github.com/analogdevicesinc/ToF/tree/master/examples/first-frame) to calculate FPS we received from ToF on NXP board.
We observed that in some random scenario the execution of our application gets stuck. We added some logs into the SDK to determine the place and found that it gets stuck inside [TofiCompute](https://github.com/analogdevicesinc/ToF/blob/master/sdk/src/cameras/itof-camera/camera_itof.cpp#L785) and never comes out of it.

We are using the pre-built libraries present in lib folder.
The following are sha256 hashes for the same:

```
analog@aditof:~/Workspace/libs$ sha256sum ./*
48faff46c51b4ddc15d5c5f8c9795810c0b9ec7b6bf0eda19ecce24f6561ea03  ./libtofi_compute.so
3338039beca91b20992175a52f47a96372be0dedf564f76abadc4fa81862faf0  ./libtofi_config.so
002b0d97b888dbf85655b976e280cf861b2b5faf8ac29f152265996beb7a09dd  ./tofi_compute_depth
```

Below is the content of config files that we are using:

config_crosby_old_modes.json 
```
{
    "VAUX_POWER_VOLTAGE": "18",
    "DEPTH_INI": "./config/RawToDepthAdsd3500_mp.ini;./config/RawToDepthAdsd3500_qmp.ini",
    "FPS": "30",
    "FSYNC_MODE": "1"
}

```

RawToDepthAdsd3500_mp.ini 
```
abThreshMin=1.0
abSumThresh=25.0
confThresh=25.0
radialThreshMin=100.0
radialThreshMax=10000.0
jblfApplyFlag=1
jblfWindowSize=5
jblfGaussianSigma=10.0
jblfExponentialTerm=5.0
jblfMaxEdge=12.0
jblfABThreshold=10.0
headerSize=0
deltaCompEnable=0
inputFormat=raw16_bits12_shift4
depthComputeIspEnable=1
partialDepthEnable=1
interleavingEnable=0
bitsInPhaseOrDepth=12
bitsInConf=0
bitsInAB=12
phaseInvalid=0
xyzEnable=1

```

RawToDepthAdsd3500_qmp.ini 
```
abThreshMin=1.0
abSumThresh=25.0
confThresh=25.0
radialThreshMin=100.0
radialThreshMax=10000.0
jblfApplyFlag=1
jblfWindowSize=5
jblfGaussianSigma=10.0
jblfExponentialTerm=5.0
jblfMaxEdge=12.0
jblfABThreshold=10.0
headerSize=0
deltaCompEnable=0
inputFormat=raw8
depthComputeIspEnable=1
partialDepthEnable=0
interleavingEnable=1
bitsInPhaseOrDepth=16
bitsInConf=8
bitsInAB=16
phaseInvalid=0
xyzEnable=1

```

Here are the power-on logs of application from the SDK:

```
I20230602 07:02:53.295928   712 system_impl.cpp:90] SDK built with websockets version:3.1.0
I20230602 07:02:53.296499   712 sensor_enumerator_imx.cpp:116] Looking for sensors on the target
I20230602 07:02:53.297179   712 sensor_enumerator_imx.cpp:137] Looking at: /dev/media0 for an eligible TOF camera
I20230602 07:02:53.312410   712 sensor_enumerator_imx.cpp:153] Considering: /dev/media0 an eligible TOF camera
I20230602 07:02:53.316262   712 camera_itof.cpp:143] Initializing camera
sh: 1: cannot create /sys/class/gpio/gpio122/value: Permission denied
sh: 1: cannot create /sys/class/gpio/gpio122/value: Permission denied
I20230602 07:03:00.422422   712 adsd3500_sensor.cpp:220] Opening device
I20230602 07:03:00.422609   712 adsd3500_sensor.cpp:238] Looking for the following cards:
I20230602 07:03:00.422658   712 adsd3500_sensor.cpp:240] mxc-isi-cap
I20230602 07:03:00.422717   712 adsd3500_sensor.cpp:252] device: /dev/video0    subdevice: /dev/v4l-subdev1
W20230602 07:03:00.438637   712 adsd3500_sensor.cpp:1417] Failed to read imager type and CCB version (command 0x0032). Possibly command is not implemented on the current adsd3500 firmware.
W20230602 07:03:00.438735   712 adsd3500_sensor.cpp:1425] Since the image type is unknown, fall back on compile flag to determine imager type
I20230602 07:03:00.447700   712 mode_info.cpp:129] Using old modes table.
I20230602 07:03:00.447890   712 mode_info.cpp:129] Using old modes table.
I20230602 07:03:00.504799   712 camera_itof.cpp:347] Current adsd3500 firmware version is: 3.1.1.0
I20230602 07:03:00.504911   712 camera_itof.cpp:350] Current adsd3500 firmware git hash is: 7e9902fd27bda6747d898db19fef7d9faf361a69
I20230602 07:03:00.508042   712 camera_itof.cpp:1716] Found Depth ini file: ./config/RawToDepthAdsd3500_mp.ini
I20230602 07:03:00.508263   712 camera_itof.cpp:1716] Found Depth ini file: ./config/RawToDepthAdsd3500_qmp.ini
I20230602 07:03:00.508327   712 camera_itof.cpp:1727] Current Depth ini file is: ./config/RawToDepthAdsd3500_mp.ini
I20230602 07:03:00.512357   712 camera_itof.cpp:441] Camera FPS set from Json file at: 30
I20230602 07:03:00.513356   712 camera_itof.cpp:453] Camera initialized
I20230602 07:03:00.513540   712 main.cpp:123] SD card image version: microsd-93635899.img
I20230602 07:03:00.513586   712 main.cpp:124] Kernel version: lf-5.10.72-2.2.0
I20230602 07:03:00.513665   712 main.cpp:125] U-Boot version: imx_v2020.04_5.4.70_2.3.0
I20230602 07:03:00.513753   712 main.cpp:135] Availble frame types:
I20230602 07:03:00.513799   712 main.cpp:138] qmp
I20230602 07:03:00.513842   712 main.cpp:138] mp
I20230602 07:03:00.519964   712 camera_itof.cpp:530] Chosen mode: qmp
I20230602 07:03:00.520061   712 camera_itof.cpp:569] Using ini file: ./config/RawToDepthAdsd3500_qmp.ini
I20230602 07:03:00.525203   712 adsd3500_sensor.cpp:220] Opening device
I20230602 07:03:00.525285   712 adsd3500_sensor.cpp:238] Looking for the following cards:
I20230602 07:03:00.525319   712 adsd3500_sensor.cpp:240] mxc-isi-cap
I20230602 07:03:00.525355   712 adsd3500_sensor.cpp:252] device: /dev/video0    subdevice: /dev/v4l-subdev1
I20230602 07:03:00.525549   712 adsd3500_sensor.cpp:428] Setting camera mode to qmp
I20230602 07:03:00.532475   712 camera_itof.cpp:913] initComputeLibrary
I20230602 07:03:00.888823   712 adsd3500_sensor.cpp:341] Starting device 0

```
Let me know if any additional information is required.

meetgandhi-eic, 2023-06-02:
Here is the output of top command:

```
analog@aditof:~$ top -H -p 623 -bn1
top - 07:49:11 up 11 min,  2 users,  load average: 0.12, 0.36, 0.22
Threads:   5 total,   0 running,   5 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.0 us,  0.0 sy,  0.0 ni,100.0 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   2690.3 total,   2117.3 free,    364.3 used,    208.7 buff/cache
MiB Swap:      0.0 total,      0.0 free,      0.0 used.   2256.5 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
    623 analog    20   0  334420  25716   8400 S   0.0   0.9   1:28.18 tof-ros-node
    628 analog    20   0  334420  25716   8400 S   0.0   0.9   0:24.39 tof-ros-node
    629 analog    20   0  334420  25716   8400 S   0.0   0.9   0:14.78 tof-ros-node
    630 analog    20   0  334420  25716   8400 S   0.0   0.9   0:14.36 tof-ros-node
    631 analog    20   0  334420  25716   8400 S   0.0   0.9   0:17.07 tof-ros-node

```

andrestraker, 2023-06-09:
Try running **data_collect** with **sudo**. From this in the log that does not seem to be the case now:

I20230602 07:02:53.316262   712 camera_itof.cpp:143] Initializing camera
**sh: 1: cannot create /sys/class/gpio/gpio122/value: Permission denied
sh: 1: cannot create /sys/class/gpio/gpio122/value: Permission denied**

andrestraker, 2023-06-09:
Also, what is the serial number of you device? I want to make sure you are using the correct JSON file.

meetgandhi-eic, 2023-06-13:
I'm not sure from where to get the serial number.
Below are some details which might help:

Model:

```
analog@aditof:~$ cat /proc/device-tree/model
NXP i.MX8MPlus ADI TOF carrier + ADSD3500
```

Serial Number from dmesg running on host:

```
[520949.222919] usb 2-4: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[520949.243309] usb 2-4: New USB device found, idVendor=0456, idProduct=a4a2, bcdDevice= 1.00
[520949.243312] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[520949.243314] usb 2-4: Product: ADI TOF USB Gadget
[520949.243315] usb 2-4: Manufacturer: Analog Devices Inc.
[520949.243316] usb 2-4: SerialNumber: 12345678

```

andrestraker, 2023-06-13:
The serial number is usually on the device.

For example:
[image]

meetgandhi-eic, 2023-06-19:
I'm still trying to look for the serial number (The sticker is not there on the device).
However, I found one instead of `DV11-163` my device has `CR-413`.

meetgandhi-eic, 2023-06-20:
[out_raw_.bin.zip](https://github.com/analogdevicesinc/ToF/files/11800150/out_raw_.bin.zip)
I tried to capture the "raw data" of the last captured frame before ToFI compute gets stuck. Attaching it here in the zip.

andrestraker, 2023-06-20:
@meetgandhi-eic Are you referring to the call in the code or the command line tool tofi compute depth? If the latter, can you also send the command line parameters you used.

meetgandhi-eic, 2023-06-20:
I'm referring to the call in the code.

andrestraker, 2023-06-20:
@meetgandhi-eic  Let's start by aligning the firmware version with SDK version. Given that you are using SDK version 4.1.1, I would suggest updating the firmware to version 4.1.5.  See the eval kit 4.1.1 release to get the firmware:
https://github.com/analogdevicesinc/ToF/releases/tag/v4.1.1

Note, version v4.2.0 eval kit has also been released.

How did you save the data that was sent?

meetgandhi-eic, 2023-06-21:
> The serial number is usually on the device.
> 
> For example: [image]

Okay. I get the mode table information from [here](https://wiki.analog.com/resources/eval/user-guides/eval-adtf3175d-nxz/mode_table)
According to the same, the current ToF serial number is starting from CR-xxx.

andrestraker, 2023-06-23:
@meetgandhi-eic  just one note, the firmware is still: 3.1.1.0

I20230621 10:21:58.366457  2142 camera_itof.cpp:355] Current adsd3500 firmware version is: 3.1.1.0

meetgandhi-eic, 2023-06-28:
> @meetgandhi-eic just one note, the firmware is still: 3.1.1.0
> 
> I20230621 10:21:58.366457 2142 camera_itof.cpp:355] Current adsd3500 firmware version is: 3.1.1.0
> 
> Please also update the SD card image to the 4.2.0 image.

Hi @andrestraker ,
I have received the ToF kit with newer version support. I tried running the application in "lr-qnative" mode. But got the same results. Here are the insights for the same for further debugging:

Application source code:

```
#include "aditof/aditof.h"
#include "aditof/sensor_enumerator_factory.h"
#include "aditof/sensor_enumerator_interface.h"
#include "aditof/camera.h"

#include <iostream>
#include <linux/videodev2.h>
#include <map>
#include <string>
#include <sys/time.h>
#include <vector>
#include <memory>
#include <chrono>
#include <fstream>
#include <glog/logging.h>
#include <thread>
#include <queue>
#include <mutex>
#include <stdlib.h>
#include <signal.h>

using std::chrono::duration_cast;
using std::chrono::milliseconds;
using std::chrono::steady_clock;
using std::chrono::time_point;
using std::chrono::time_point_cast;

using TimePoint = time_point<steady_clock, milliseconds>;
static inline TimePoint current_timepoint()
{
    return (time_point_cast<milliseconds>(steady_clock::now()));
}

/* Available sensors */
using namespace aditof;
aditof::System tofSystem;
std::vector<std::shared_ptr<aditof::Camera>> cameras;
std::shared_ptr<aditof::Camera> depthCam;
std::queue<aditof::Frame *> framequeue;
std::mutex m_mtxQueue;
bool sensorStarted = false;
bool captureTimedout = false;

Status save_frame(aditof::Frame &frame, std::string frameType)
{

    uint16_t *data1;
    FrameDataDetails fDetails;
    Status status = Status::OK;

    status = frame.getData(frameType, &data1);
    if (status != Status::OK)
    {
        LOG(ERROR) << "Could not get frame data " + frameType + "!";
        return status;
    }

    if (!data1)
[... 280 lines cut ...]

add_executable(tof-ros-node)

link_directories(/home/einfochips/data/cam_sync/libs)

FILE(GLOB SRC_FILES *.cpp *.hpp *.c *.h)
target_sources(tof-ros-node PRIVATE ${SRC_FILES})

find_package(aditof REQUIRED)
find_package(Libwebsockets REQUIRED)
find_package(glog REQUIRED)
find_package(Protobuf REQUIRED)

target_include_directories(tof-ros-node PUBLIC "${ADITOF_INCLUDE_DIRS}")
target_include_directories(tof-ros-node PUBLIC "${LIBWEBSOCKETS_INCLUDE_DIRS}")
target_link_libraries(tof-ros-node "${ADITOF_LIBRARIES}")
target_link_libraries(tof-ros-node aditof)
target_link_libraries(tof-ros-node /home/analog/Workspace/libs/libtofi_compute.so)
target_link_libraries(tof-ros-node /home/analog/Workspace/libs/libtofi_config.so)
target_link_libraries(tof-ros-node websockets_shared)
target_link_libraries(tof-ros-node glog::glog)
target_link_libraries(tof-ros-node pthread)
```

Let me know if you need any additional infomration.

closed 2024-04-26 by andrestraker