analogdevicesinc/ToF #349: Failed to get frame in UVC mode on NVIDIA AGX Orin

meetgandhi-eic, 2023-06-14:
Hi,

I'm trying to capture frames on NVIDIA AGX Orin over UVC mode.
I have done following changes in the NXP i.MX8 image:

1. Enabled UVC mode  by updating [usb-gadget.sh](https://github.com/analogdevicesinc/ToF/blob/master/sdcard-images-utils/nxp/patches/ubuntu_overlay/step1/usr/share/systemd/usb-gadget.sh#L231)
```
 create_uvc configs/c.1 uvc.0
 #create_rndis configs/c.1 rndis.0
 #create_msd configs/c.1 mass_storage.0 /dev/mmcblk1p1
```
2. uvc-gadget.sh gets run on boot by default which runs the uvc-app.

I'm able to see the camera gets probed with NVIDIA board as UVC gadget:
```
^[[A[Jun14 15:52] usb 2-3.2: new SuperSpeed Gen 1 USB device number 4 using tegra-xusb
[  +0.026913] usb 2-3.2: New USB device found, idVendor=0456, idProduct=a4a2, bcdDevice= 1.00
[  +0.000012] usb 2-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000005] usb 2-3.2: Product: ADI TOF USB Gadget
[  +0.000003] usb 2-3.2: Manufacturer: Analog Devices Inc.
[  +0.000003] usb 2-3.2: SerialNumber: 12345678
[  +0.011682] uvcvideo: Found UVC 1.00 device ADI TOF USB Gadget (0456:a4a2)
[  +0.016394] input: ADI TOF USB Gadget: UVC Camera as /devices/platform/3610000.xhci/usb2/2-3/2-3.2/2-3.2:1.0/input/input6
```

Now, I'm trying to capture the frame using first-frame application but it gets failed with following logs:

```
orin-5@orin-5:~/tof_sdk_v4.2.0/ToF/build/examples/first-frame$ ./first-frame config/config_crosby_old_modes.json 
I20230614 15:32:11.824067  3075 main.cpp:78] SDK version: 4.1.1 | branch: HEAD | commit: a223f75
I20230614 15:32:11.824874  3075 system_impl.cpp:90] SDK built with websockets version:3.1.0
I20230614 15:32:11.824997  3075 usb_sensor_enumerator.cpp:61] Looking for USB connected sensors
W20230614 15:32:11.825412  3075 usb_sensor_enumerator.cpp:116] /dev/video1 xioctl failed' error: 22(Invalid argument)
I20230614 15:32:11.825747  3075 usb_sensor_enumerator.cpp:148] Found USB capture device at: /dev/video0
I20230614 15:32:11.843737  3075 usb_sensor_enumerator.cpp:181] Received the following message with available sensors from target: sensors_info {
  image_sensors {
    name: "adsd3500"
  }
}
card_image_version {
  kernelversion: "lf-5.10.72-2.2.0"
  sdversion: "microsd-4.2.0-b260ad46.img"
  ubootversion: "imx_v2020.04_5.4.70_2.3.0"
}
W20230614 15:32:11.847643  3075 usb_sensor_enumerator.cpp:116] /dev/nvidia-modeset xioctl failed' error: 25(Inappropriate ioctl for device)
I20230614 15:32:11.848603  3075 camera_itof.cpp:143] Initializing camera
I20230614 15:32:11.848634  3075 usb_depth_sensor_linux.cpp:122] Opening device
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

After some debugging, I came to know that the bad_alloc exception is coming from [here](https://github.com/analogdevicesinc/ToF/blob/master/sdk/src/connections/usb/linux/usb_linux_utils.cpp#L360). The application is trying to reserve a large buffer.

To conclude the same, I have handled that exception to see the buffer size (changes can be found in attached uvc-gadget-changes-v4.2.0.patch).
It gave below log:

```
orin-5@orin-5:~/tof_sdk_v4.2.0/ToF/build/examples/first-frame$ ./first-frame config/config_crosby_old_modes.json 
I20230614 15:35:45.837957  3345 main.cpp:78] SDK version: 4.1.1 | branch: HEAD | commit: a223f75
I20230614 15:35:45.838637  3345 system_impl.cpp:90] SDK built with websockets version:3.1.0
I20230614 15:35:45.838748  3345 usb_sensor_enumerator.cpp:61] Looking for USB connected sensors
W20230614 15:35:45.839125  3345 usb_sensor_enumerator.cpp:116] /dev/video1 xioctl failed' error: 22(Invalid argument)
I20230614 15:35:46.044770  3345 usb_sensor_enumerator.cpp:148] Found USB capture device at: /dev/video0
I20230614 15:35:46.063216  3345 usb_sensor_enumerator.cpp:181] Received the following message with available sensors from target: sensors_info {
  image_sensors {
    name: "adsd3500"
  }
}
card_image_version {
  kernelversion: "lf-5.10.72-2.2.0"
  sdversion: "microsd-4.2.0-b260ad46.img"
  ubootversion: "imx_v2020.04_5.4.70_2.3.0"
}
W20230614 15:35:46.072944  3345 usb_sensor_enumerator.cpp:116] /dev/nvidia-modeset xioctl failed' error: 25(Inappropriate ioctl for device)
I20230614 15:35:46.075507  3345 camera_itof.cpp:143] Initializing camera
I20230614 15:35:46.075582  3345 usb_depth_sensor_linux.cpp:122] Opening device
W20230614 15:35:46.090349  3345 usb_linux_utils.cpp:365] bad_alloc caught: std::bad_alloc
W20230614 15:35:46.090481  3345 usb_linux_utils.cpp:366] Request length: 187651734896643
E20230614 15:35:46.090569  3345 usb_depth_sensor_linux.cpp:160] Response for get available frame types request failed
W20230614 15:35:46.091068  3345 camera_itof.cpp:150] Failed to open device
E20230614 15:35:46.091140  3345 main.cpp:110] Could not initialize camera!
```

It seems like some garbage length is coming which is causing this issue.

Here are the changes that I have done on ToF SDK v4.2.0 running on NVIDIA.

```
orin-5@orin-5:~/tof_sdk_v4.2.0/ToF$ git diff 
diff --git a/examples/first-frame/main.cpp b/examples/first-frame/main.cpp
index ef5cd6e..ca1c18a 100644
--- a/examples/first-frame/main.cpp
+++ b/examples/first-frame/main.cpp
@@ -124,7 +124,7 @@ int main(int argc, char *argv[]) {
         std::cout << "no frame type avaialble!";
         return 0;
     }
-    status = camera->setFrameType("lr-qnative");
+    status = camera->setFrameType(frameTypes.at(0));
     if (status != Status::OK) {
         LOG(ERROR) << "Could not set camera frame type!";
         return 0;
diff --git a/sdk/src/connections/usb/linux/usb_linux_utils.cpp b/sdk/src/connections/usb/linux/usb_linux_utils.cpp
index f52e6e1..82a40c5 100644
--- a/sdk/src/connections/usb/linux/usb_linux_utils.cpp
+++ b/sdk/src/connections/usb/linux/usb_linux_utils.cpp
@@ -357,7 +357,15 @@ aditof::Status UsbLinuxUtils::uvcExUnitGetResponse(int fd,
     }
     size_t stringLength = reinterpret_cast<size_t *>(packet)[0];
 
-    responseStr.reserve(stringLength);
+    try {
+        responseStr.reserve(stringLength);
+    }
+    catch (std::bad_alloc& ba)
+    {
+           LOG(WARNING) << "bad_alloc caught: " << ba.what();
+           LOG(WARNING) << "Request length: " << stringLength;
+           return aditof::Status::GENERIC_ERROR;
+    }
 
     size_t readBytes = 0;
     size_t readlength = 0;
diff --git a/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp b/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
index 567c432..2a27eb5 100644
--- a/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
+++ b/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
@@ -113,14 +113,19 @@ Status UsbSensorEnumerator::searchSensors() {
         CLEAR(fmt);
         fmt.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
         if (-1 == UsbLinuxUtils::xioctl(fd, VIDIOC_G_FMT, &fmt)) {
+                LOG(WARNING) << driverPath << " xioctl failed"
+                                        << "' error: " << errno << "(" << strerror(errno)
+                         << ")";
+
             close(fd);
             continue;
         }
 
-        std::string devName("ADI CMOS TOF UVC Gadget");
+        std::string devName("ADI TOF USB Gadget: UVC Camera");
 
         if (strncmp(reinterpret_cast<const char *>(cap.card), devName.c_str(),
                     devName.length()) != 0) {
+                LOG(WARNING) << driverPath << " ignoring " << reinterpret_cast<const char *>(cap.card);
             close(fd);
             continue;
         }
(END)

```

Let me know if you need any additional information.

andrestraker, 2023-06-14:
@meetgandhi-eic  - we are investigating this issue. We will get back to you soon.

meetgandhi-eic, 2023-06-15:
Just for an Update:
- I switched to sdk v4.2.0, but still facing the same issue:

```
orin-5@orin-5:~/tof_sdk_v4.2.0/ToF/build/examples/first-frame$ ./first-frame config/config_crosby_old_modes.json 
I20230615 11:13:29.361694  7308 main.cpp:78] SDK version: 4.2.0 | branch: HEAD | commit: b260ad46
I20230615 11:13:29.362488  7308 system_impl.cpp:90] SDK built with websockets version:3.1.0
I20230615 11:13:29.362610  7308 usb_sensor_enumerator.cpp:61] Looking for USB connected sensors
W20230615 11:13:29.363013  7308 usb_sensor_enumerator.cpp:116] /dev/video1 xioctl with VIDIOC_G_FMT failed' error: 22(Invalid argument)
I20230615 11:13:29.574553  7308 usb_sensor_enumerator.cpp:147] Found USB capture device at: /dev/video0
I20230615 11:13:29.601634  7308 usb_sensor_enumerator.cpp:180] Received the following message with available sensors from target: sensors_info {
  image_sensors {
    name: "adsd3500"
  }
}
card_image_version {
  kernelversion: "lf-5.10.72-2.2.0"
  sdversion: "microsd-4.2.0-b260ad46.img"
  ubootversion: "imx_v2020.04_5.4.70_2.3.0"
}
W20230615 11:13:29.611668  7308 usb_sensor_enumerator.cpp:116] /dev/nvidia-modeset xioctl with VIDIOC_G_FMT failed' error: 25(Inappropriate ioctl for device)
I20230615 11:13:29.614054  7308 camera_itof.cpp:146] Initializing camera
I20230615 11:13:29.614140  7308 usb_depth_sensor_linux.cpp:122] Opening device
W20230615 11:13:29.628187  7308 usb_linux_utils.cpp:365] bad_alloc caught: std::bad_alloc
W20230615 11:13:29.628304  7308 usb_linux_utils.cpp:366] Request length: 187650443051011
E20230615 11:13:29.628401  7308 usb_depth_sensor_linux.cpp:160] Response for get available frame types request failed
W20230615 11:13:29.628995  7308 camera_itof.cpp:153] Failed to open device
E20230615 11:13:29.629065  7308 main.cpp:110] Could not initialize camera!
```

The below patch is applied on the sdk:

```
diff --git a/examples/first-frame/main.cpp b/examples/first-frame/main.cpp
index ef5cd6ea..ca1c18a6 100644
--- a/examples/first-frame/main.cpp
+++ b/examples/first-frame/main.cpp
@@ -124,7 +124,7 @@ int main(int argc, char *argv[]) {
         std::cout << "no frame type avaialble!";
         return 0;
     }
-    status = camera->setFrameType("lr-qnative");
+    status = camera->setFrameType(frameTypes.at(0));
     if (status != Status::OK) {
         LOG(ERROR) << "Could not set camera frame type!";
         return 0;
diff --git a/sdk/src/connections/usb/linux/usb_linux_utils.cpp b/sdk/src/connections/usb/linux/usb_linux_utils.cpp
index f52e6e1b..82a40c5b 100644
--- a/sdk/src/connections/usb/linux/usb_linux_utils.cpp
+++ b/sdk/src/connections/usb/linux/usb_linux_utils.cpp
diff --git a/examples/first-frame/main.cpp b/examples/first-frame/main.cpp
index ef5cd6ea..ca1c18a6 100644
--- a/examples/first-frame/main.cpp
+++ b/examples/first-frame/main.cpp
@@ -124,7 +124,7 @@ int main(int argc, char *argv[]) {
         std::cout << "no frame type avaialble!";
         return 0;
     }
-    status = camera->setFrameType("lr-qnative");
+    status = camera->setFrameType(frameTypes.at(0));
     if (status != Status::OK) {
         LOG(ERROR) << "Could not set camera frame type!";
         return 0;
diff --git a/sdk/src/connections/usb/linux/usb_linux_utils.cpp b/sdk/src/connections/usb/linux/usb_linux_utils.cpp
index f52e6e1b..82a40c5b 100644
--- a/sdk/src/connections/usb/linux/usb_linux_utils.cpp
+++ b/sdk/src/connections/usb/linux/usb_linux_utils.cpp
@@ -357,7 +357,15 @@ aditof::Status UsbLinuxUtils::uvcExUnitGetResponse(int fd,
     }
     size_t stringLength = reinterpret_cast<size_t *>(packet)[0];
 
-    responseStr.reserve(stringLength);
+    try {
+        responseStr.reserve(stringLength);
+    }
+    catch (std::bad_alloc& ba)
+    {
+           LOG(WARNING) << "bad_alloc caught: " << ba.what();
+           LOG(WARNING) << "Request length: " << stringLength;
+           return aditof::Status::GENERIC_ERROR;
+    }
 
     size_t readBytes = 0;
     size_t readlength = 0;
diff --git a/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp b/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
index acaea720..4278ec30 100644
--- a/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
+++ b/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
@@ -113,6 +113,9 @@ Status UsbSensorEnumerator::searchSensors() {
         CLEAR(fmt);
         fmt.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
         if (-1 == UsbLinuxUtils::xioctl(fd, VIDIOC_G_FMT, &fmt)) {
+                LOG(WARNING) << driverPath << " xioctl with VIDIOC_G_FMT failed"
+                                        << "' error: " << errno << "(" << strerror(errno)
+                         << ")";
             close(fd);
             continue;
         }
@@ -121,6 +124,7 @@ Status UsbSensorEnumerator::searchSensors() {
 
         if (strncmp(reinterpret_cast<const char *>(cap.card), devName.c_str(),
                     devName.length()) != 0) {
+                LOG(WARNING) << driverPath << " ignoring " << reinterpret_cast<const char *>(cap.card);
             close(fd);
             continue;
         }

```

andrestraker, 2023-06-15:
@meetgandhi-eic  did you intend to attach the log to the message?

meetgandhi-eic, 2023-06-15:
With the UVC mode, the application is not reaching at the point where it can request the available frame types from the camera. The execution gets failed [here](https://github.com/analogdevicesinc/ToF/blob/v4.2.0/examples/first-frame/main.cpp#L108) in the initialize method.

 W20230615 18:20:40.266026  8026 usb_linux_utils.cpp:365] bad_alloc caught: std::bad_alloc
W20230615 18:20:40.266156  8026 usb_linux_utils.cpp:366] Request length: 187650896035843
E20230615 18:20:40.266250  8026 usb_depth_sensor_linux.cpp:160] Response for get available frame types request failed
**W20230615 18:20:40.266803  8026 camera_itof.cpp:153] Failed to open device**
**E20230615 18:20:40.266871  8026 main.cpp:110] Could not initialize camera!**

meetgandhi-eic, 2023-06-15:
yep. It works pretty well in the network mode. We haven't observed any issues so far.
Than you for the support.

SeptimiuVana, 2023-06-15:
Hello,

You can keep these change or you can specify which frame type you would like to use by passing a string
with the name of the frame type.
`status = camera->setFrameType(frameTypes.at(0));`

You also have to change this line from:
https://github.com/analogdevicesinc/ToF/blob/v4.2.0/sdk/src/connections/usb/linux/usb_linux_utils.cpp#L358

`size_t stringLength = reinterpret_cast<size_t *>(packet)[0];`

to 

`size_t stringLength = size_t(packet[0]);`

It should work fine.

NOTE: please run the first frame example twice because at the first run it does some configuration on the nxp and it will return an error.

meetgandhi-eic, 2023-06-16:
@SeptimiuVana 

The changes you suggested seems to be working. I was able to capture the frames by running the application multiple times as mentioned.

Here are the logs coming from the same.

```
orin-5@orin-5:~/tof_sdk_v4.2.0/ToF/build/examples/first-frame$ ./first-frame config/config_crosby_old_modes.json 
I20230616 11:52:57.949363  4305 main.cpp:78] SDK version: 4.2.0 | branch: HEAD | commit: b260ad46
I20230616 11:52:57.950124  4305 system_impl.cpp:90] SDK built with websockets version:3.1.0
I20230616 11:52:57.950258  4305 usb_sensor_enumerator.cpp:61] Looking for USB connected sensors
W20230616 11:52:57.950744  4305 usb_sensor_enumerator.cpp:116] /dev/nvidia-modeset xioctl with VIDIOC_G_FMT failed' error: 25(Inappropriate ioctl for device)
W20230616 11:52:57.951092  4305 usb_sensor_enumerator.cpp:116] /dev/video1 xioctl with VIDIOC_G_FMT failed' error: 22(Invalid argument)
I20230616 11:52:58.158118  4305 usb_sensor_enumerator.cpp:147] Found USB capture device at: /dev/video0
I20230616 11:52:58.183038  4305 usb_sensor_enumerator.cpp:180] Received the following message with available sensors from target: sensors_info {
  image_sensors {
    name: "adsd3500"
  }
}
card_image_version {
  kernelversion: "lf-5.10.72-2.2.0"
  sdversion: "microsd-4.2.0-b260ad46.img"
  ubootversion: "imx_v2020.04_5.4.70_2.3.0"
}
I20230616 11:52:58.195495  4305 camera_itof.cpp:146] Initializing camera
I20230616 11:52:58.195641  4305 usb_depth_sensor_linux.cpp:122] Opening device
I20230616 11:52:58.305694  4305 mode_info.cpp:156] Using old modes table.
I20230616 11:52:58.372807  4305 camera_itof.cpp:355] Current adsd3500 firmware version is: 3.1.1.0
I20230616 11:52:58.372879  4305 camera_itof.cpp:357] Current adsd3500 firmware git hash is: 7e9902fd27bda6747d898db19fef7d9faf361a69
I20230616 11:52:58.373245  4305 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3500_mp.ini
I20230616 11:52:58.373378  4305 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3500_qmp.ini
I20230616 11:52:58.373426  4305 camera_itof.cpp:1777] Current Depth ini file is: ./config/RawToDepthAdsd3500_mp.ini
I20230616 11:52:58.373888  4305 camera_itof.cpp:450] Camera FPS set from Json file at: 10
I20230616 11:52:58.390605  4305 camera_itof.cpp:462] Camera initialized
I20230616 11:52:58.390781  4305 main.cpp:117] SD card image version: microsd-4.2.0-b260ad46.img
I20230616 11:52:58.390806  4305 main.cpp:118] Kernel version: lf-5.10.72-2.2.0
I20230616 11:52:58.390825  4305 main.cpp:119] U-Boot version: imx_v2020.04_5.4.70_2.3.0
no frame type avaialble!orin-5@orin-5:~/tof_sdk_v4.2.0/ToF/build/examples/first-frame$ ./first-frame config/config_crosby_old_modes.json 
I20230616 11:53:02.146497  4307 main.cpp:78] SDK version: 4.2.0 | branch: HEAD | commit: b260ad46
I20230616 11:53:02.147269  4307 system_impl.cpp:90] SDK built with websockets version:3.1.0
I20230616 11:53:02.147405  4307 usb_sensor_enumerator.cpp:61] Looking for USB connected sensors
W20230616 11:53:02.147792  4307 usb_sensor_enumerator.cpp:116] /dev/nvidia-modeset xioctl with VIDIOC_G_FMT failed' error: 25(Inappropriate ioctl for device)
W20230616 11:53:02.148088  4307 usb_sensor_enumerator.cpp:116] /dev/video1 xioctl with VIDIOC_G_FMT failed' error: 22(Invalid argument)
I20230616 11:53:02.354377  4307 usb_sensor_enumerator.cpp:147] Found USB capture device at: /dev/video0
I20230616 11:53:02.376099  4307 usb_sensor_enumerator.cpp:180] Received the following message with available sensors from target: sensors_info {
  image_sensors {
    name: "adsd3500"
  }
}
card_image_version {
  kernelversion: "lf-5.10.72-2.2.0"
  sdversion: "microsd-4.2.0-b260ad46.img"
  ubootversion: "imx_v2020.04_5.4.70_2.3.0"
}
I20230616 11:53:02.387720  4307 camera_itof.cpp:146] Initializing camera
I20230616 11:53:02.387815  4307 usb_depth_sensor_linux.cpp:122] Opening device
I20230616 11:53:02.465893  4307 mode_info.cpp:156] Using old modes table.
I20230616 11:53:02.714216  4307 camera_itof.cpp:355] Current adsd3500 firmware version is: 3.1.1.0
I20230616 11:53:02.714308  4307 camera_itof.cpp:357] Current adsd3500 firmware git hash is: 7e9902fd27bda6747d898db19fef7d9faf361a69
I20230616 11:53:02.714830  4307 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3500_mp.ini
I20230616 11:53:02.715055  4307 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3500_qmp.ini
I20230616 11:53:02.715126  4307 camera_itof.cpp:1777] Current Depth ini file is: ./config/RawToDepthAdsd3500_mp.ini
I20230616 11:53:02.715888  4307 camera_itof.cpp:450] Camera FPS set from Json file at: 10
I20230616 11:53:02.730224  4307 camera_itof.cpp:462] Camera initialized
I20230616 11:53:02.730405  4307 main.cpp:117] SD card image version: microsd-4.2.0-b260ad46.img
I20230616 11:53:02.730448  4307 main.cpp:118] Kernel version: lf-5.10.72-2.2.0
I20230616 11:53:02.730477  4307 main.cpp:119] U-Boot version: imx_v2020.04_5.4.70_2.3.0
Available Frame types:
qmp
mp
[... 93 lines cut ...]
index acaea720..4278ec30 100644
--- a/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
+++ b/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
@@ -113,6 +113,9 @@ Status UsbSensorEnumerator::searchSensors() {
         CLEAR(fmt);
         fmt.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
         if (-1 == UsbLinuxUtils::xioctl(fd, VIDIOC_G_FMT, &fmt)) {
+                LOG(WARNING) << driverPath << " xioctl with VIDIOC_G_FMT failed"
[... cut ...]
             close(fd);
             continue;
         }
@@ -121,6 +124,7 @@ Status UsbSensorEnumerator::searchSensors() {
 
         if (strncmp(reinterpret_cast<const char *>(cap.card), devName.c_str(),
                     devName.length()) != 0) {
+                LOG(WARNING) << driverPath << " ignoring " << reinterpret_cast<const char *>(cap.card);
             close(fd);
             continue;
         }
```

I'm exploring this further to capture multiple frames.
Thank you for the support.

meetgandhi-eic, 2023-06-16:
Update:

I tried to interface the ToF in UVC mode on an x86_64 host. It seems to be working without any modifications in `sdk/src/connections/usb/linux/usb_linux_utils.cpp`. However, as @SeptimiuVana mentioned it fails in the first attempt but then works as exepected.

Here is the patch for the change in sdk (only first-frame logic is changed to select "qmp" mode and capturing images continuously to calculate FPS).

```
diff --git a/examples/first-frame/main.cpp b/examples/first-frame/main.cpp
index ef5cd6ea..2d53e0c1 100644
--- a/examples/first-frame/main.cpp
+++ b/examples/first-frame/main.cpp
@@ -42,6 +42,19 @@
 #include <ios>
 #include <iostream>
 
+#include <chrono>
+
+using std::chrono::time_point_cast;
+using std::chrono::time_point;
+using std::chrono::steady_clock;
+using std::chrono::milliseconds;
+using std::chrono::duration_cast;
+
+using TimePoint = time_point<steady_clock, milliseconds>;
+static inline TimePoint current_timepoint() {
+           return (time_point_cast<milliseconds>(steady_clock::now()));
+}
+
 using namespace aditof;
 
 Status save_frame(aditof::Frame &frame, std::string frameType) {
@@ -73,7 +86,7 @@ Status save_frame(aditof::Frame &frame, std::string frameType) {
 int main(int argc, char *argv[]) {
 
     google::InitGoogleLogging(argv[0]);
-    FLAGS_alsologtostderr = 1;
+    FLAGS_alsologtostderr = 0;
 
     LOG(INFO) << "SDK version: " << aditof::getApiVersion()
               << " | branch: " << aditof::getBranchVersion()
@@ -124,7 +137,13 @@ int main(int argc, char *argv[]) {
         std::cout << "no frame type avaialble!";
         return 0;
     }
-    status = camera->setFrameType("lr-qnative");
+
+    std::cout << "Availble frame types" << std::endl;
+    for (auto item : frameTypes) {
+        std::cout  << item << std::endl;
+    }
+
+    status = camera->setFrameType(frameTypes.at(0));
     if (status != Status::OK) {
         LOG(ERROR) << "Could not set camera frame type!";
         return 0;
@@ -136,34 +155,65 @@ int main(int argc, char *argv[]) {
         return 0;
     }
     aditof::Frame frame;
+    uint32_t nFrames = 0;
+    auto fpsTimeStamp = current_timepoint();
+    bool dumpFirst = false;
 
-
+    while (1) {
     status = camera->requestFrame(&frame);
     if (status != Status::OK) {
         LOG(ERROR) << "Could not request frame!";
         return 0;
     } else {
[... 117 lines cut ...]
Linux orin-5 5.10.104-tegra #1 SMP PREEMPT Wed Aug 10 20:17:07 PDT 2022 aarch64 aarch64 aarch64 GNU/Linux
```

- UVC Driver version:

x86_64:
```
meet@AHMLPT1819:~$ modinfo uvcvideo
filename:       /lib/modules/5.4.0-125-generic/updates/dkms/uvcvideo.ko
version:        1.1.2.realsense-1.3.18
license:        GPL
description:    USB Video Class driver
author:         Laurent Pinchart <laurent.pinchart@ideasonboard.com>
srcversion:     51A4A0210E91BE50A7BA2BB
```
AGX Orin:
```
orin-5@orin-5:~/tof_sdk_v4.2.0/ToF/build/examples/first-frame$ modinfo uvcvideo
filename:       /lib/modules/5.10.104-tegra/kernel/drivers/media/usb/uvc/uvcvideo.ko
version:        1.1.1
license:        GPL
description:    USB Video Class driver
author:         Laurent Pinchart <laurent.pinchart@ideasonboard.com>
srcversion:     187DBDC4F94931600634224
```

SeptimiuVana, 2023-06-16:
Hello,

Thank you for the detailed setup description.

You are running an older firmware (inside the camera) that doesn't support temperature reading.
Can you comment this 2 lines out of the example?(first-frame)

https://github.com/analogdevicesinc/ToF/blob/master/examples/first-frame/main.cpp#L155
https://github.com/analogdevicesinc/ToF/blob/master/examples/first-frame/main.cpp#L160

Please reboot the nxp board before running the new code.

meetgandhi-eic, 2023-06-19:
> @meetgandhi-eic you're welcome. Please note, we are deprecating the modules with old modes (QMP, MP). Please contact the FAE working with you to get the number of modules you need supporting the new modes (lr-native, lr-qnative, etc.).

@andrestraker can you explain more what changes will be required to support the new modes? is it something we can manage with the firmware or any hardware level changes will be required?

meetgandhi-eic, 2023-06-19:
> Hello,
> 
> Thank you for the detailed setup description.
> 
> You are running an older firmware (inside the camera) that doesn't support temperature reading. Can you comment this 2 lines out of the example?(first-frame)
> 
> https://github.com/analogdevicesinc/ToF/blob/master/examples/first-frame/main.cpp#L155 https://github.com/analogdevicesinc/ToF/blob/master/examples/first-frame/main.cpp#L160
> 
> Please reboot the nxp board before running the new code.
> 
> You should be able to get multiple frames now.

Hi @SeptimiuVana,
I tried that but still no luck.
Looking it the below log, seems like it is not able to capture the frame.

**W20230619 11:16:45.094807  3202 usb_depth_sensor_linux.cpp:507] select timeout: 
W20230619 11:16:45.095036  3202 camera_itof.cpp:743] Failed to get frame from device**

Here's the details log of running the application multiple times:

```
orin-5@orin-5:~/tof_sdk_v4.2.0/ToF/build/examples/first-frame$ ./first-frame config/config_crosby_old_modes.json 
I20230619 11:16:31.126307  3200 main.cpp:91] SDK version: 4.2.0 | branch: HEAD | commit: b260ad46
I20230619 11:16:31.126850  3200 system_impl.cpp:90] SDK built with websockets version:3.1.0
I20230619 11:16:31.126966  3200 usb_sensor_enumerator.cpp:61] Looking for USB connected sensors
W20230619 11:16:31.127269  3200 usb_sensor_enumerator.cpp:116] /dev/nvidia-modeset xioctl with VIDIOC_G_FMT failed' error: 25(Inappropriate ioctl for device)
W20230619 11:16:31.127485  3200 usb_sensor_enumerator.cpp:116] /dev/video1 xioctl with VIDIOC_G_FMT failed' error: 22(Invalid argument)
I20230619 11:16:31.334712  3200 usb_sensor_enumerator.cpp:147] Found USB capture device at: /dev/video0
I20230619 11:16:31.360793  3200 usb_sensor_enumerator.cpp:180] Received the following message with available sensors from target: sensors_info {
  image_sensors {
    name: "adsd3500"
  }
}
card_image_version {
  kernelversion: "lf-5.10.72-2.2.0"
  sdversion: "microsd-4.2.0-b260ad46.img"
  ubootversion: "imx_v2020.04_5.4.70_2.3.0"
}
I20230619 11:16:31.368693  3200 camera_itof.cpp:146] Initializing camera
I20230619 11:16:31.368788  3200 usb_depth_sensor_linux.cpp:122] Opening device
I20230619 11:16:31.488977  3200 mode_info.cpp:156] Using old modes table.
I20230619 11:16:31.545722  3200 camera_itof.cpp:355] Current adsd3500 firmware version is: 3.1.1.0
I20230619 11:16:31.545787  3200 camera_itof.cpp:357] Current adsd3500 firmware git hash is: 7e9902fd27bda6747d898db19fef7d9faf361a69
I20230619 11:16:31.546064  3200 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3500_mp.ini
I20230619 11:16:31.546169  3200 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3500_qmp.ini
I20230619 11:16:31.546212  3200 camera_itof.cpp:1777] Current Depth ini file is: ./config/RawToDepthAdsd3500_mp.ini
I20230619 11:16:31.546574  3200 camera_itof.cpp:450] Camera FPS set from Json file at: 30
I20230619 11:16:31.560333  3200 camera_itof.cpp:462] Camera initialized
I20230619 11:16:31.560425  3200 main.cpp:130] SD card image version: microsd-4.2.0-b260ad46.img
I20230619 11:16:31.560460  3200 main.cpp:131] Kernel version: lf-5.10.72-2.2.0
I20230619 11:16:31.560482  3200 main.cpp:132] U-Boot version: imx_v2020.04_5.4.70_2.3.0
no frame type avaialble!orin-5@orin-5:~/tof_sdk_v4.2.0/ToF/build/examples/first-frame$ ./first-frame config/config_crosby_old_modes.json 
I20230619 11:16:33.733350  3202 main.cpp:91] SDK version: 4.2.0 | branch: HEAD | commit: b260ad46
I20230619 11:16:33.744940  3202 system_impl.cpp:90] SDK built with websockets version:3.1.0
I20230619 11:16:33.745029  3202 usb_sensor_enumerator.cpp:61] Looking for USB connected sensors
W20230619 11:16:33.745316  3202 usb_sensor_enumerator.cpp:116] /dev/nvidia-modeset xioctl with VIDIOC_G_FMT failed' error: 25(Inappropriate ioctl for device)
W20230619 11:16:33.745530  3202 usb_sensor_enumerator.cpp:116] /dev/video1 xioctl with VIDIOC_G_FMT failed' error: 22(Invalid argument)
I20230619 11:16:33.950755  3202 usb_sensor_enumerator.cpp:147] Found USB capture device at: /dev/video0
I20230619 11:16:33.968127  3202 usb_sensor_enumerator.cpp:180] Received the following message with available sensors from target: sensors_info {
  image_sensors {
    name: "adsd3500"
  }
}
card_image_version {
  kernelversion: "lf-5.10.72-2.2.0"
  sdversion: "microsd-4.2.0-b260ad46.img"
  ubootversion: "imx_v2020.04_5.4.70_2.3.0"
}
I20230619 11:16:33.973964  3202 camera_itof.cpp:146] Initializing camera
[... 161 lines cut ...]
diff --git a/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp b/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
index acaea720..4278ec30 100644
--- a/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
+++ b/sdk/src/connections/usb/linux/usb_sensor_enumerator.cpp
@@ -113,6 +113,9 @@ Status UsbSensorEnumerator::searchSensors() {
         CLEAR(fmt);
         fmt.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
         if (-1 == UsbLinuxUtils::xioctl(fd, VIDIOC_G_FMT, &fmt)) {
+                LOG(WARNING) << driverPath << " xioctl with VIDIOC_G_FMT failed"
+                                        << "' error: " << errno << "(" << strerror(errno)
+                         << ")";
             close(fd);
             continue;
         }
@@ -121,6 +124,7 @@ Status UsbSensorEnumerator::searchSensors() {
 
         if (strncmp(reinterpret_cast<const char *>(cap.card), devName.c_str(),
                     devName.length()) != 0) {
+                LOG(WARNING) << driverPath << " ignoring " << reinterpret_cast<const char *>(cap.card);
             close(fd);
             continue;
         }
```

Also, the API _requestFrame_ is returning success even though the intenal API `status = m_depthSensor->getFrame(frameDataLocation);` has returned error here: https://github.com/analogdevicesinc/ToF/blob/v4.2.0/sdk/src/cameras/itof-camera/camera_itof.cpp#L743

andrestraker, 2023-06-23:
Thank you @meetgandhi-eic that definitely needs a return on failure instead of just logging the message. We will fix that shortly.

https://github.com/analogdevicesinc/ToF/blob/master/sdk/src/cameras/itof-camera/camera_itof.cpp#L738

andrestraker, 2023-06-23:
> @andrestraker can you explain more what changes will be required to support the new modes? is it something we can manage with the firmware or any hardware level changes will be required?

Besides having a module with the new modes you really will not have to do anything other than use the new modes as you would the old modes with the API calls. Just ensure you have the latest updates for the SDK on the host and the device. And the latest Linux driver, etc., on the device. If you are connecting the ADTF317D to the host then update the SD to the version that comes with 4.2.0.

meetgandhi-eic, 2023-06-23:
> Thank you @meetgandhi-eic that definitely needs a return on failure instead of just logging the message. We will fix that shortly.
> 
> https://github.com/analogdevicesinc/ToF/blob/master/sdk/src/cameras/itof-camera/camera_itof.cpp#L738

> Besides having a module with the new modes you really will not have to do anything other than use the new modes as you would the old modes with the API calls. Just ensure you have the latest updates [...]

No. I mean what can be done in the current kit/sensor (starting with serial number CR-XXX to support the new modes? Or it will no longer work with the future releases of SDK?

andrestraker, 2023-06-23:
From the 4.3.0 eval kit release (SDK 4.3.0) old modes will be deprecated.

Contact the FAE you work with to get an updated module to use with your current eval kit.

closed 2024-04-26 by andrestraker