analogdevicesinc/ToF #402: Python3 ToF Bindings Seg Fault on NXP

pclass-sensonix, 2023-08-22:
Hello - I am trying to use the Python3 ToF bindings on the NXP i.MX8. I can successfully capture frames from the ADSD3500 Camera with V4L2 and the following script on the i.MX8:
```

v4l2-ctl --set-ctrl=operating_mode=0 -d /dev/v4l-subdev1
v4l2-ctl --set-ctrl=phase_depth_bits=4 -d /dev/v4l-subdev1
v4l2-ctl --set-ctrl=ab_bits=6 -d /dev/v4l-subdev1
v4l2-ctl --set-ctrl=confidence_bits=0 -d /dev/v4l-subdev1
v4l2-ctl --set-ctrl=ab_averaging=0 -d /dev/v4l-subdev1
v4l2-ctl --set-ctrl=depth_enable=0 -d /dev/v4l-subdev1
v4l2-ctl --device /dev/video0 --set-fmt-video=width=2048,height=2560,pixelformat=BA81 --stream-mmap --stream-to=mode0.bin --stream-count=$nr_frames
--
[  673.848452] bypass csc
[  673.848462] input fmt BA81
[  673.848464] output fmt BA81
[  674.262957] isi-capture 32e00000.isi:cap_device: ISI channel[0] has stopped
```

However, when I try to run the python3 example script below:

```
import aditofpython as tof
import numpy as np

system = tof.System()

cameras = []
status = system.getCameraList(cameras)
print("system.getCameraList()", status)

camera1 = cameras[0]

modes = []
status = camera1.getAvailableModes(modes)
print("system.getAvailableModes()", status)
print(modes)

camera1.setControl("initialization_config", "./tof-viewer_config.json")
status = camera1.initialize()
print("camera1.initialize()", status)

types = []
status = camera1.getAvailableFrameTypes(types)
print("system.getAvailableFrameTypes()", status)
print(types)

camDetails = tof.CameraDetails()
status = camera1.getDetails(camDetails)
print("system.getDetails()", status)
print("camera1 details:", "id:", camDetails.cameraId, "connection:", camDetails.connection)

status = camera1.setFrameType(types[0])
print("camera1.setFrameType()", status)

status = camera1.setMode(modes[0])
print("camera1.setMode()", status)
```

I get the following error:

```
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0324 10:38:33.923457   658 system_impl.cpp:90] SDK built with websockets version:4.1.6-v4.1.6
I0324 10:38:33.923589   658 sensor_enumerator_imx.cpp:116] Looking for sensors on the target
I0324 10:38:33.923907   658 sensor_enumerator_imx.cpp:137] Looking at: /dev/media0 for an eligible TOF camera
I0324 10:38:33.964524   658 sensor_enumerator_imx.cpp:153] Considering: /dev/media0 an eligible TOF camera
system.getCameraList() Status.Ok
system.getAvailableModes() Status.Ok
['pcmmp', 'srqmp', 'lrqmp', 'srmp', 'lrmp']
I0324 10:38:33.965276   658 camera_itof.cpp:146] Initializing camera
I0324 10:38:33.965310   658 adsd3500_sensor.cpp:207] Opening device
I0324 10:38:33.965328   658 adsd3500_sensor.cpp:225] Looking for the following cards:
I0324 10:38:33.965343   658 adsd3500_sensor.cpp:227] mxc-isi-cap
I0324 10:38:33.965356   658 adsd3500_sensor.cpp:239] device: /dev/video0	subdevice: /dev/v4l-subdev1
W0324 10:38:33.966929   658 adsd3500_sensor.cpp:990] Reading Adsd3500 error errno: 5 error: Input/output error
I0324 10:38:39.818653   658 adsd3500_interrupt_notifier.cpp:33] Received signal 0 from kernel
I0324 10:38:39.822163   658 adsd3500_sensor.cpp:1554] statusRegister:0(Adsd3500Status::OK)
I0324 10:38:39.822237   658 adsd3500_interrupt_notifier.cpp:33] Received signal 0 from kernel
I0324 10:38:39.825736   658 adsd3500_sensor.cpp:1554] statusRegister:0(Adsd3500Status::OK)
I0324 10:38:39.838490   658 adsd3500_sensor.cpp:335] ADSD3500 is ready to communicate with.
I0324 10:38:39.852028   658 mode_info.cpp:148] Using new modes table for adsd3500.
I0324 10:38:39.860694   658 mode_info.cpp:144] Using new mixed modes table for adsd3500.
I0324 10:38:39.860736   658 mode_info.cpp:144] Using new mixed modes table for adsd3500.
I0324 10:38:40.028141   658 camera_itof.cpp:355] Current adsd3500 firmware version is: 4.2.1.0
I0324 10:38:40.028206   658 camera_itof.cpp:357] Current adsd3500 firmware git hash is: a59afad6d6c8e7fb153a2db8cd8840359f17195a
I0324 10:38:40.032168   658 camera_itof.cpp:1777] Current Depth ini file is: ./RawToDepthAdsd3500_sr-native.ini
I0324 10:38:40.034569   658 camera_itof.cpp:450] Camera FPS set from Json file at: 10
I0324 10:38:40.035817   658 camera_itof.cpp:462] Camera initialized
camera1.initialize() Status.Ok
system.getAvailableFrameTypes() Status.Ok
['sr-native', 'lr-native', 'sr-qnative', 'lr-qnative', 'pcm-native', 'sr-mixed', 'lr-mixed']
system.getDetails() Status.Ok
camera1 details: id:  connection: ConnectionType.???
I0324 10:38:40.041918   658 camera_itof.cpp:539] Chosen mode: sr-native
I0324 10:38:40.041970   658 camera_itof.cpp:593] Using ini file: ./RawToDepthAdsd3500_sr-native.ini
I0324 10:38:40.047096   658 adsd3500_sensor.cpp:207] Opening device
I0324 10:38:40.047125   658 adsd3500_sensor.cpp:225] Looking for the following cards:
I0324 10:38:40.047142   658 adsd3500_sensor.cpp:227] mxc-isi-cap
I0324 10:38:40.047160   658 adsd3500_sensor.cpp:239] device: /dev/video0	subdevice: /dev/v4l-subdev1
I0324 10:38:40.047346   658 adsd3500_sensor.cpp:463] Setting camera mode to sr-native
I0324 10:38:40.058508   658 camera_itof.cpp:924] initComputeLibrary
Segmentation fault
```

The working directory looks like this:

```
ls -rlt
total 20
-rwxr-x---    1 root     root           912 Mar 24 10:26 test.py
-rw-r-----    1 root     root           117 Mar 24 10:28 tof-viewer_config.json
-rw-r-----    1 root     root           406 Mar 24 10:30 RawToDepthAdsd3500_sr-native.ini
-rw-r-----    1 root     root          8136 Mar 24 10:32 libtofi_compute.so
```

The config files look like this:

root@imx8-gateway:/test# cat tof-viewer_config.json
{
"skip_network_cameras": "off",
"DEPTH_INI": "./RawToDepthAdsd3500_sr-native.ini",
"FPS": "10",
"FSYNC_MODE": "1"
}
root@imx8-gateway:/test# cat RawToDepthAdsd3500_sr-native.ini 
abThreshMin=1.0
abSumThresh=25.0
confThresh=25.0
radialThreshMin=30.0
radialThreshMax=4200.0
jblfApplyFlag=1
jblfWindowSize=5
jblfGaussianSigma=10.0
jblfExponentialTerm=5.0
jblfMaxEdge=12.0
jblfABThreshold=10.0
headerSize=0
deltaCompEnable=0
inputFormat=mipiRaw12_8
depthComputeIspEnable=0
partialDepthEnable=0
interleavingEnable=0
bitsInPhaseOrDepth=12
bitsInConf=0
bitsInAB=16
phaseInvalid=0
xyzEnable=1

pclass-sensonix, 2023-08-22:
The strace command gives this information:

```
write(2, "I0324 10:33:35.942015   670 adsd"..., 86I0324 10:33:35.942015   670 adsd3500_sensor.cpp:463] Setting camera mode to sr-native
) = 86
ioctl(5, VIDIOC_S_CTRL, {id=V4L2_CTRL_CLASS_USER+0x19e0, value=0 => 0}) = 0
ioctl(4, VIDIOC_REQBUFS, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, memory=V4L2_MEMORY_MMAP, count=0 => 0}) = 0
ioctl(4, VIDIOC_S_FMT, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, fmt.pix={width=2048, height=2560, pixelformat=v4l2_fourcc('B', 'A', '8', '1') /* V4L2_PIX_FMT_SBGGR8 */, field=V4L2_FIELD_ANY, bytesperline=0, sizeimage=0, colorspace=V4L2_COLORSPACE_DEFAULT}} => {fmt.pix={width=2048, height=2560, pixelformat=v4l2_fourcc('B', 'A', '8', '1') /* V4L2_PIX_FMT_SBGGR8 */, field=V4L2_FIELD_NONE, bytesperline=2048, sizeimage=5242880, colorspace=V4L2_COLORSPACE_SRGB}}) = 0
ioctl(4, VIDIOC_REQBUFS, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, memory=V4L2_MEMORY_MMAP, count=4 => 4}) = 0
ioctl(4, VIDIOC_QUERYBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=0, memory=V4L2_MEMORY_MMAP, m.offset=0, length=5242880, bytesused=0, flags=V4L2_BUF_FLAG_TIMESTAMP_MONOTONIC|V4L2_BUF_FLAG_TSTAMP_SRC_EOF, ...}) = 0
mmap(NULL, 5242880, PROT_READ|PROT_WRITE, MAP_SHARED, 4, 0) = 0xffff914a2000
ioctl(4, VIDIOC_QUERYBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=1, memory=V4L2_MEMORY_MMAP, m.offset=0x500000, length=5242880, bytesused=0, flags=V4L2_BUF_FLAG_TIMESTAMP_MONOTONIC|V4L2_BUF_FLAG_TSTAMP_SRC_EOF, ...}) = 0
mmap(NULL, 5242880, PROT_READ|PROT_WRITE, MAP_SHARED, 4, 0x500000) = 0xffff90fa2000
ioctl(4, VIDIOC_QUERYBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=2, memory=V4L2_MEMORY_MMAP, m.offset=0xa00000, length=5242880, bytesused=0, flags=V4L2_BUF_FLAG_TIMESTAMP_MONOTONIC|V4L2_BUF_FLAG_TSTAMP_SRC_EOF, ...}) = 0
mmap(NULL, 5242880, PROT_READ|PROT_WRITE, MAP_SHARED, 4, 0xa00000) = 0xffff90aa2000
ioctl(4, VIDIOC_QUERYBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=3, memory=V4L2_MEMORY_MMAP, m.offset=0xf00000, length=5242880, bytesused=0, flags=V4L2_BUF_FLAG_TIMESTAMP_MONOTONIC|V4L2_BUF_FLAG_TSTAMP_SRC_EOF, ...}) = 0
mmap(NULL, 5242880, PROT_READ|PROT_WRITE, MAP_SHARED, 4, 0xf00000) = 0xffff905a2000
gettid()                                = 670
write(2, "I0324 10:33:35.967603   670 came"..., 68I0324 10:33:35.967603   670 camera_itof.cpp:924] initComputeLibrary
) = 68
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x88e60bfabe104} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

pclass-sensonix, 2023-08-22:
Clinfo shows the OpenCL library and support as well:

```
/usr/bin/camera# clinfo
Number of platforms                               1
  Platform Name                                   Vivante OpenCL Platform
  Platform Vendor                                 Vivante Corporation
  Platform Version                                OpenCL 3.0 V6.4.3.p2.336687
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_byte_addressable_store cl_khr_fp16
  Platform Host timer resolution                  0ns
```

SeptimiuVana, 2023-08-23:
Hello, 

Please go to the build folder and check if you have the USE_DEPTH_COMPUTE_STUBS flag enabled. You can do this by running cmake -LA in  the build folder.

If it's set to 1/on you need to set it to off and rebuild the project.
Before you rebuild the project you need to copy the libtofi_compute.so and libtofi_config.so from the installer onto the nxp
in a folder called libs right next to the /home/analog/Workspace/ToF folder.

After you copy the libraries go to the build folder and run:
cmake -DUSE_DEPTH_COMPUTE_STUBS=0 -DWITH_PYTHON=1 ..
make -j4

You can get all the files required to run a python script if you go to the build/bindings/python/examples/first_frame folder.
You need the libaditof.so* files, the libtofi_config.so, libtofi_compute.so and the config folder.

pclass-sensonix, 2023-08-24:
Hi - thanks for the input.

I made sure to make my directory structure look like the first_frame example. Now I am getting a segmentation fault in a different spot.

Here is the output of the first_frame script:

```
python3 ./first_frame.py ./config.json 
Looking for camera on UVC. Will use ./tof-viewer_config.json.
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0324 11:04:33.411160   801 system_impl.cpp:90] SDK built with websockets version:4.1.6-v4.1.6
I0324 11:04:33.411204   801 sensor_enumerator_imx.cpp:116] Looking for sensors on the target
I0324 11:04:33.411525   801 sensor_enumerator_imx.cpp:137] Looking at: /dev/media0 for an eligible TOF camera
I0324 11:04:33.438766   801 sensor_enumerator_imx.cpp:153] Considering: /dev/media0 an eligible TOF camera
system.getCameraList() Status.Ok
camera1.setControl() Status.Ok
I0324 11:04:33.439373   801 camera_itof.cpp:146] Initializing camera
I0324 11:04:33.439400   801 adsd3500_sensor.cpp:207] Opening device
I0324 11:04:33.439416   801 adsd3500_sensor.cpp:225] Looking for the following cards:
I0324 11:04:33.439429   801 adsd3500_sensor.cpp:227] mxc-isi-cap
I0324 11:04:33.439442   801 adsd3500_sensor.cpp:239] device: /dev/video0        subdevice: /dev/v4l-subdev1
I0324 11:04:33.449576   801 adsd3500_sensor.cpp:335] ADSD3500 is ready to communicate with.
I0324 11:04:33.461863   801 mode_info.cpp:148] Using new modes table for adsd3500.
I0324 11:04:33.471195   801 mode_info.cpp:144] Using new mixed modes table for adsd3500.
I0324 11:04:33.471231   801 mode_info.cpp:144] Using new mixed modes table for adsd3500.
I0324 11:04:33.830659   801 camera_itof.cpp:355] Current adsd3500 firmware version is: 4.2.1.0
I0324 11:04:33.830741   801 camera_itof.cpp:357] Current adsd3500 firmware git hash is: a59afad6d6c8e7fb153a2db8cd8840359f17195a
I0324 11:04:33.830858   801 camera_itof.cpp:1833] Current Depth ini file is: RawToDepthAdsd3500_lr-qnative.ini
I0324 11:04:33.832592   801 camera_itof.cpp:450] Camera FPS set from Json file at: 10
I0324 11:04:33.833477   801 camera_itof.cpp:462] Camera initialized
camera1.initialize() Status.Ok
camera1.getAvailableFrameTypes() Status.Ok
['sr-native', 'lr-native', 'sr-qnative', 'lr-qnative', 'pcm-native', 'sr-mixed', 'lr-mixed']
camera1.getDetails() Status.Ok
camera1 details: id:  connection: ConnectionType.???
I0324 11:04:33.839605   801 camera_itof.cpp:539] Chosen mode: lr-qnative
I0324 11:04:33.839637   801 camera_itof.cpp:593] Using ini file: RawToDepthAdsd3500_lr-qnative.ini
I0324 11:04:33.844734   801 adsd3500_sensor.cpp:207] Opening device
I0324 11:04:33.844756   801 adsd3500_sensor.cpp:225] Looking for the following cards:
I0324 11:04:33.844772   801 adsd3500_sensor.cpp:227] mxc-isi-cap
I0324 11:04:33.844790   801 adsd3500_sensor.cpp:239] device: /dev/video0        subdevice: /dev/v4l-subdev1
I0324 11:04:33.845203   801 adsd3500_sensor.cpp:463] Setting camera mode to lr-qnative
I0324 11:04:33.855988   801 camera_itof.cpp:924] initComputeLibrary
Segmentation fault
```

The line that it seg faults on is this one:

```
                 m_tofi_config = InitTofiConfig_isp((ConfigFileData *)&depth_ini,
                                                    convertedMode, &status,
                                                    m_xyz_dealias_data);
```

My config.json file looks like this:

```
cat ./tof-viewer_config.json 
{
    "VAUX_POWER_VOLTAGE": "18",
    "skip_network_cameras": "off",
    "camera_ip": "10.42.0.1",
    "DEPTH_INI": "RawToDepthAdsd3500_lr-qnative.ini",
    "FPS": "10",
    "FSYNC_MODE": "1"
}
```

Can you please help me figure out what is wrong? Thanks.

pclass-sensonix, 2023-08-24:
I did make sure that **USE_DEPTH_COMPUTE_STUBS** was off.

```
grep -R "USE_DEPTH_COMPUTE_STUBS" *
:USE_DEPTH_COMPUTE_STUBS:BOOL=offCMakeCache.txt:USE_DEPTH_COMPUTE_STUBS:BOOL=off
```

pclass-sensonix, 2023-08-24:
Here is the directory structure that I have:

```
root@imx8-gateway:/usr/bin/camera/test# ls -rlt
total 38580
-rw-r-----    1 root     root           457 Mar 24 10:31 tof_tembin_adsd3500_new_modes_config.json
-rw-r-----    1 root     root           457 Mar 24 10:31 tof_crosby_adsd3500_new_modes_config.json
drwxr-x---    2 root     root          4096 Mar 24 10:38 config
-rwxr-xr-x    1 root     root           912 Mar 24 10:42 python-test.py
-rwxr-x---    1 root     root          1899 Mar 24 10:45 first_frame.py
-rw-r-----    1 root     root           405 Mar 24 10:48 RawToDepthAdsd3500_sr-native.ini
-rw-r-----    1 root     root           399 Mar 24 10:49 RawToDepthAdsd3500_lr-qnative.ini
-rw-r-----    1 root     root           195 Mar 24 11:10 config.json
-rw-r-----    1 root     root      13151608 Mar 24 11:12 libaditof.so.4.2.0
-rw-r-----    1 root     root      13151608 Mar 24 11:12 libaditof.so.1.0
-rw-r-----    1 root     root      13151608 Mar 24 11:12 libaditof.so
-rw-r-----    1 root     root          8192 Mar 24 11:12 libtofi_config.so
-rw-r-----    1 root     root          8136 Mar 24 11:12 libtofi_compute.so
```

Here is the configs directory:

```
root@imx8-gateway:/usr/bin/camera/test# ls -R config 
config:
RawToDepthAdsd3030_lr-mixed.ini    RawToDepthAdsd3030_sr-mixed.ini    RawToDepthAdsd3500_lr-mixed.ini    RawToDepthAdsd3500_mp.ini          RawToDepthAdsd3500_sr-native.ini
RawToDepthAdsd3030_lr-native.ini   RawToDepthAdsd3030_sr-native.ini   RawToDepthAdsd3500_lr-native.ini   RawToDepthAdsd3500_qmp.ini         RawToDepthAdsd3500_sr-qnative.ini
RawToDepthAdsd3030_lr-qnative.ini  RawToDepthAdsd3030_sr-qnative.ini  RawToDepthAdsd3500_lr-qnative.ini  RawToDepthAdsd3500_sr-mixed.ini    RawToDepthAdsd_pcm-native.ini
```

pclass-sensonix, 2023-08-25:
It does not seem to matter what the camera mode is. If I change the mode, the ini is found, but the seg fault still occurs.

```
I0324 10:28:59.139642   604 camera_itof.cpp:1833] Current Depth ini file is: ./RawToDepthAdsd3500_mp.ini
I0324 10:28:59.141372   604 camera_itof.cpp:450] Camera FPS set from Json file at: 10
I0324 10:28:59.142261   604 camera_itof.cpp:462] Camera initialized
camera1.initialize() Status.Ok
camera1.getAvailableFrameTypes() Status.Ok
['sr-native', 'lr-native', 'sr-qnative', 'lr-qnative', 'pcm-native', 'sr-mixed', 'lr-mixed']
camera1.getDetails() Status.Ok
camera1 details: id:  connection: ConnectionType.???
I0324 10:28:59.148275   604 camera_itof.cpp:539] Chosen mode: lr-qnative
I0324 10:28:59.148306   604 camera_itof.cpp:593] Using ini file: ./RawToDepthAdsd3500_mp.ini
I0324 10:28:59.153419   604 adsd3500_sensor.cpp:207] Opening device
I0324 10:28:59.153445   604 adsd3500_sensor.cpp:225] Looking for the following cards:
I0324 10:28:59.153465   604 adsd3500_sensor.cpp:227] mxc-isi-cap
I0324 10:28:59.153484   604 adsd3500_sensor.cpp:239] device: /dev/video0        subdevice: /dev/v4l-subdev1
I0324 10:28:59.153669   604 adsd3500_sensor.cpp:463] Setting camera mode to lr-qnative
I0324 10:28:59.167840   604 camera_itof.cpp:924] initComputeLibrary
I0324 10:28:59.167896   604 camera_itof.cpp:931] here 1
I0324 10:28:59.167907   604 camera_itof.cpp:938] here 2
I0324 10:28:59.167918   604 camera_itof.cpp:942] here 3
I0324 10:28:59.167929   604 camera_itof.cpp:949] here 4
I0324 10:28:59.167943   604 camera_itof.cpp:959] here 5
I0324 10:28:59.167956   604 camera_itof.cpp:965] here 6
I0324 10:28:59.167968   604 camera_itof.cpp:969] here 7
Segmentation fault
```

pclass-sensonix, 2023-08-25:
I have done another test. I downloaded your official NXP image, burned it onto a flash card, mounted that flash card, and extracted the following two directories:

```
/home/analog/Workspace/ToF/build/examples
/home/analog/Workspace/ToF/bindings/python/examples/first_frame
```

I copied these wholesale to my NXP dev kit. It doesn't matter if I run the Python bindings for 'first-frame' or the C++ version. They both seg fault.

```
root@imx8-gateway:/home/first_frame# ls
CMakeFiles                                    aditofpython.cpython-38-aarch64-linux-gnu.so  common                                        libaditof.so.1.0                              libtofi_config.so
Makefile                                      buffer.pb.cc                                  config                                        libaditof.so.4.2.0                            libwebsockets.so.14
aditof-config-version.cmake                   buffer.pb.h                                   first_frame.py                                libglog.so.1                                  usb_buffer.pb.cc
aditof-config.cmake                           cmake_install.cmake                           libaditof.so                                  libtofi_compute.so                            usb_buffer.pb.h
root@banner-fusion-imx8-gateway:/home/first_frame# python3 ./first_frame.py ./config/config_adsd3500_adsd3030_new_modes.json
WARNING: Logging before InitGoogleLogging() is written to STDERR
I20210324 13:20:43.599526  1513 system_impl.cpp:90] SDK built with websockets version:3.1.0
I20210324 13:20:43.599650  1513 sensor_enumerator_imx.cpp:116] Looking for sensors on the target
I20210324 13:20:43.600034  1513 sensor_enumerator_imx.cpp:137] Looking at: /dev/media0 for an eligible TOF camera
I20210324 13:20:43.609023  1513 sensor_enumerator_imx.cpp:153] Considering: /dev/media0 an eligible TOF camera
...
write(2, "I20210324 13:23:00.930301  1545 "..., 72I20210324 13:23:00.930301  1545 camera_itof.cpp:924] initComputeLibrary
) = 72
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_ACCERR, si_addr=0xffffa93ff778} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

I only have the official libraries distributed in the official 4.2 release on the NXP. 

Any ideas why this is seg faulting?

pclass-sensonix, 2023-08-25:
I used the official ADTF3175D - Release v4.2.0 image and flashed it to a new sd card. I booted the device and it still seg faults. Could the camera be causing this? The camera seems fine because I can take frames with v4l2-ctl, but I cannot use the ADI sdk:

```
analog@aditof:~/Workspace/ToF/build/examples/first-frame$ ./first-frame config/config_adsd3500_adsd3030_new_modes.json
I20230315 15:07:35.372468   582 main.cpp:78] SDK version: 4.2.0 | branch: HEAD | commit: b260ad46
I20230315 15:07:35.373270   582 system_impl.cpp:90] SDK built with websockets version:3.1.0
I20230315 15:07:35.373414   582 sensor_enumerator_imx.cpp:116] Looking for sensors on the target
I20230315 15:07:35.373764   582 sensor_enumerator_imx.cpp:137] Looking at: /dev/media0 for an eligible TOF camera
I20230315 15:07:35.379109   582 sensor_enumerator_imx.cpp:153] Considering: /dev/media0 an eligible TOF camera
W20230315 15:07:35.380399   582 adsd3500_interrupt_notifier.cpp:57] Failed to open the debug sysfs. Interrupts support will not be available!
I20230315 15:07:35.381754   582 camera_itof.cpp:146] Initializing camera
I20230315 15:07:35.381847   582 adsd3500_sensor.cpp:207] Opening device
I20230315 15:07:35.381908   582 adsd3500_sensor.cpp:225] Looking for the following cards:
I20230315 15:07:35.381948   582 adsd3500_sensor.cpp:227] mxc-isi-cap
I20230315 15:07:35.381997   582 adsd3500_sensor.cpp:239] device: /dev/video0    subdevice: /dev/v4l-subdev1
I20230315 15:07:35.391999   582 adsd3500_sensor.cpp:335] ADSD3500 is ready to communicate with.
I20230315 15:07:35.403398   582 mode_info.cpp:148] Using new modes table for adsd3500.
I20230315 15:07:35.412307   582 mode_info.cpp:144] Using new mixed modes table for adsd3500.
I20230315 15:07:35.412420   582 mode_info.cpp:144] Using new mixed modes table for adsd3500.
I20230315 15:07:35.579528   582 camera_itof.cpp:355] Current adsd3500 firmware version is: 4.2.1.0
I20230315 15:07:35.579658   582 camera_itof.cpp:357] Current adsd3500 firmware git hash is: a59afad6d6c8e7fb153a2db8cd8840359f17195a
I20230315 15:07:35.580209   582 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3030_sr-native.ini
I20230315 15:07:35.580366   582 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3030_lr-native.ini
I20230315 15:07:35.580433   582 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3030_sr-qnative.ini
I20230315 15:07:35.580550   582 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3030_lr-qnative.ini
I20230315 15:07:35.580608   582 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3030_sr-mixed.ini
I20230315 15:07:35.580670   582 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd3030_lr-mixed.ini
I20230315 15:07:35.580730   582 camera_itof.cpp:1766] Found Depth ini file: ./config/RawToDepthAdsd_pcm-native.ini
I20230315 15:07:35.580799   582 camera_itof.cpp:1777] Current Depth ini file is: ./config/RawToDepthAdsd3030_lr-mixed.ini
I20230315 15:07:35.583379   582 camera_itof.cpp:450] Camera FPS set from Json file at: 10
I20230315 15:07:35.584336   582 camera_itof.cpp:462] Camera initialized
I20230315 15:07:35.584461   582 main.cpp:117] SD card image version: microsd-4.2.0-b260ad46.img
I20230315 15:07:35.584515   582 main.cpp:118] Kernel version: lf-5.10.72-2.2.0
I20230315 15:07:35.584565   582 main.cpp:119] U-Boot version: imx_v2020.04_5.4.70_2.3.0
I20230315 15:07:35.590389   582 camera_itof.cpp:539] Chosen mode: lr-qnative
I20230315 15:07:35.590471   582 camera_itof.cpp:593] Using ini file: ./config/RawToDepthAdsd3030_lr-qnative.ini
I20230315 15:07:35.595618   582 adsd3500_sensor.cpp:207] Opening device
I20230315 15:07:35.595687   582 adsd3500_sensor.cpp:225] Looking for the following cards:
I20230315 15:07:35.595726   582 adsd3500_sensor.cpp:227] mxc-isi-cap
I20230315 15:07:35.595767   582 adsd3500_sensor.cpp:239] device: /dev/video0    subdevice: /dev/v4l-subdev1
I20230315 15:07:35.595984   582 adsd3500_sensor.cpp:463] Setting camera mode to lr-qnative
I20230315 15:07:35.602766   582 camera_itof.cpp:924] initComputeLibrary
Segmentation fault
analog@aditof:~/Workspace/ToF/build/examples/first-frame$
```

SeptimiuVana, 2023-08-28:
Hello, the nxp image contains by default those depth compute stub libraries. They are used only for building purposes. That's the cause of the seg fault.

Inside the install package, right next to the ADTF3175D - Release v4.2.0 image file there should be a folder called depth_compute.
In that folder you will find an installer for the depth compute library that you want to use. Please get the libtofi_compute.so and libtofi_config.so from there.

pclass-sensonix, 2023-08-28:
Hello @SeptimiuVana, thanks for the response. I have installed the depth_compute prebuilt libraries. They are the only ones under /usr/lib on the NXP and it still seg faults. I have pasted the md5sums below. Please see the screenshot also.

[image]

```
root@imx8-gateway:/# find . -name "libtofi*" | xargs md5sum
7519bde29e55ba3fa91a112c1f3279b8  ./usr/lib/libtofi_config.so
da70340cc2504ebc376dec82ee06dc32  ./usr/lib/libtofi_compute.so
root@imx8-gateway:/# 
```

Test:

```
I0324 10:37:46.168463   663 adsd3500_sensor.cpp:227] mxc-isi-cap
I0324 10:37:46.168510   663 adsd3500_sensor.cpp:239] device: /dev/video0        subdevice: /dev/v4l-subdev1
I0324 10:37:46.168715   663 adsd3500_sensor.cpp:463] Setting camera mode to sr-native
I0324 10:37:46.179436   663 camera_itof.cpp:924] initComputeLibrary
Segmentation fault
```

Is there anything else that could be missing?

pclass-sensonix, 2023-08-28:
I tried the ones from PointCloud Float as well. Also, here is the strace:

```
gettid()                                = 669
write(2, "I0324 10:39:44.547245   669 came"..., 68I0324 10:39:44.547245   669 camera_itof.cpp:924] initComputeLibrary
) = 68
mmap(NULL, 4198400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8d471000
mmap(NULL, 4198400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8d070000
mmap(NULL, 4198400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8cc6f000
mmap(NULL, 4198400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8c86e000
mmap(NULL, 4198400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8c46d000
mmap(NULL, 4198400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8c06c000
mmap(NULL, 4198400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8bc6b000
mmap(NULL, 4198400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8b86a000
mmap(NULL, 4198400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8b469000
munmap(0xffff8c86e000, 4198400)         = 0
munmap(0xffff8c46d000, 4198400)         = 0
munmap(0xffff8c06c000, 4198400)         = 0
brk(0xaaab03bbb000)                     = 0xaaab03bbb000
brk(0xaaab03dbb000)                     = 0xaaab03dbb000
brk(0xaaab041bb000)                     = 0xaaab041bb000
mmap(NULL, 6295552, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8c66e000
brk(0xaaab041eb000)                     = 0xaaab041eb000
brk(0xaaab045eb000)                     = 0xaaab045eb000
brk(0xaaab049eb000)                     = 0xaaab049eb000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

pclass-sensonix, 2023-08-28:
Hello @dNechita - 

I cleared the kernel log with 'dmesg -c' and then ran the app the create the seg fault. Nothing of note was logged to dmesg when it failed.

```
root@imx8-gateway:/first-frame# ./first-frame config/config_adsd3500_adsd3100_new_modes.json
I20210324 10:31:17.829505   676 main.cpp:78] SDK version: 4.2.0 | branch: HEAD | commit: b260ad46
I20210324 10:31:17.830018   676 system_impl.cpp:90] SDK built with websockets version:4.1.6-v4.1.6
I20210324 10:31:17.830049   676 sensor_enumerator_imx.cpp:116] Looking for sensors on the target
I20210324 10:31:17.830374   676 sensor_enumerator_imx.cpp:137] Looking at: /dev/media0 for an eligible TOF camera
I20210324 10:31:17.838143   676 sensor_enumerator_imx.cpp:153] Considering: /dev/media0 an eligible TOF camera
Segmentation fault
root@banner-fusion-imx8-gateway:/first-frame# dmesg
[  370.520536] audit: type=1701 audit(1616581877.832:8): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=676 comm="first-frame" exe="/first-frame/first-frame" sig=11 res=1
```

Here is the output from gdb also. It claims no debugging symbols:

```
root@imx8-gateway:/first-frame# gdb --args ./first-frame config/config_adsd3500_adsd3100_new_modes.json
GNU gdb (GDB) 10.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-poky-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./first-frame...
(No debugging symbols found in ./first-frame)
(gdb) 
```

SeptimiuVana, 2023-08-30:
Hello,

Can you please check the library links of the first frame c++ example and the aditof.so file and send us the output?
You can run ldd ./first-frame and ldd aditof.so just to make sure the libraries are properly linked.

andrestraker, 2024-04-26:
No response from user.