analogdevicesinc/ToF #695: XYZ bin file generation using tofi_compute_depth

Vignesh-E, 2024-06-26:
Hi there,

We are trying to extract point cloud data from the raw frame, like wise you guys did using TOF module.

`./tofi_compute_depth --I=./ADTF3175_TOF/raw_frames_release_mode3 --O=./ADTF3175_TOF/raw_frames_release_mode3/processed --CCB=./crXXX.ccb --MODE=3 --INI=config/RawToDepthAdsd3500_lr-qnative.ini --ISP_Enable=1`

if we run the above command, it will gives us three different frames namely AB, Radian depth and xyz bin files right. whats the logic used to extract xyz from raw frame.

I tried using a GenerateXYZTables() function, but still not able extract same xyz frames. 

Pls, help with necessary info.

andrestraker, 2024-07-02:
Hi,

For lr-qnative the frame received from the ADSD3500 already needs to be de-interleaved, followed by extraction of the AB, depth and confidence frames. 

Afterwards, the XYZ can be derived.

This  can be seen in the following code: https://github.com/analogdevicesinc/ToF/tree/main/sdk/common/adi/depth-compute-opensource/src

Note, we have an example that will show this functionality more directly. This example is expected to be ready by 7/3. I will update here.

Regards,
Andre

andrestraker, 2024-07-03:
No, it is not possible to do so. It is supplied as a binary in the eval kits, 4.3.0 and earlier.

Vignesh-E, 2024-07-18:
how to calcuate distance using xyz bin file?

andrestraker, 2024-07-18:
z is the distance in mm.

Vignesh-E, 2024-07-18:
This distance is from camera right?

andrestraker, 2024-07-26:
Correct.

Vignesh-E, 2024-08-06:
Hi  andre,

I'm running in mode 3, so every 4 ,5 byte has Absolute Brightness.
For few pixel it has values only in 4th byte and 5th byte is 0, for few pixel values in 4th and 5th byte

I'm reading and displaying it as uint16 data, for pixel which has data in 4th and 5th was not giving proper video.

I'm not sure how to process it for preview and display

Can you please help me on this

Thanks
Vignesh

andrestraker, 2024-08-06:
Hi Vignesh,

I am assuming you are referring to the fact that the data coming from the eval kit is interleaved before processing. The NXP we use in the eval kit does not support virtual channels. Therefore the data comes interleaved from the ToF sensors.

You can see how to de-interleave the data via the following code:
https://github.com/analogdevicesinc/ToF/blob/08d887e82b781f0e95bbc8abca724f03437b4a12/sdk/common/adi/depth-compute-opensource/src/tofiCompute.cpp#L66 

Note, this discussion is relevant to pre-v5.0.0.

Regards,
Andre

vasurk, 2025-02-27:
Hi Andrestraker,

We have been successfully configuring the ToF sensor using tof_configure_force MODE_1 for long-range 1MP mode. However, we need clarification on the expected resolution output for this mode.

In Mode 3, we used to get a resolution of 2560×512.
In Mode 1 (Long Range 1MP), the function writes 0x2026 to register 0xDA01, and we would like to confirm what resolution should be expected in this configuration.

Regards,
Vasudev R K

vasurk, 2025-02-28:
Hi Andre,

Thank you for providing the script for Mode 1.

In Mode 3, the byte arrangement for each pixel in the raw data is as follows:

Bytes 0 and 1 → Depth information
Byte 2 → Confidence data
Bytes 3 and 4 → Absolute Brightness (AB) information
Could you please confirm the byte arrangement for each pixel in Mode 1 (Long Range 1MP)? we would like to understand:

What data is stored per pixel in Mode 1?
How many bytes represent depth, confidence, and AB information?
This will help us correctly interpret and process the captured frames.

Looking forward to your response.

Best regards,
Vasudev R K

andrestraker, 2025-02-28:
Hi, 

Modes 0 and 1 (MP) require our closed source depth compute library for processing. We do not support processing of these mode otherwise.

However, to understand you needs better please contact tof@analog.com. 

Regards,
Andre

vasurk, 2025-05-21:
> z is the distance in mm.

Hi Andre,

In XYZ bin file,the arrangement of XYZ will be per pixel or first all the X co-ordinates then all the Y coordinates and later all the Z coordinates will be given?