analogdevicesinc/ToF #713: Can't connect to the camera

kmisiarz, 2024-07-16:
Hi

I'm trying to use different releases but my error is the same (tried 5.0, 4.3 3.2 and 2.1). 
Both running first_frame.py from v3.2 and v2.1 and ADIToFGui from v3.2 I'm getting error:
```
~ poetry run python first_frame.py 
WARNING: Logging before InitGoogleLogging() is written to STDERR
I20240716 15:11:18.058589 1158898 system_impl.cpp:73] SDK version: 2.1.0 | branch: HEAD | commit: ba81ccb9
I20240716 15:11:18.058822 1158898 usb_sensor_enumerator.cpp:56] Looking for USB connected sensors
I20240716 15:11:18.059065 1158898 usb_sensor_enumerator.cpp:144] Found USB capture device at: /dev/video0
W20240716 15:11:18.061036 1158898 usb_linux_utils.cpp:346] Failed to read the length of the response string. Error: -1(Unknown error -1)
```

with first_frame.py from v4.3 I'm getting:
```
poetry run python first_frame.py config/config_adsd3500_adsd3030.json
Looking for camera on UVC. Will use config/config_adsd3500_adsd3030.json.
WARNING: Logging before InitGoogleLogging() is written to STDERR
I20240716 15:41:40.577112 1165506 system_impl.cpp:92] SDK built with websockets version:3.1.0
I20240716 15:41:40.577172 1165506 usb_sensor_enumerator.cpp:61] Looking for USB connected sensors
system.getCameraList() Status.Ok
Traceback (most recent call last):
  File ".../ToF4.3/bindings/python/examples/first_frame/first_frame.py", line 62, in <module>
    camera1 = cameras[0]
IndexError: list index out of range
```

I'm using ubuntu 22.04 with kernel version 6.5, python3.10
I've build the SDK 4.3 using the sh file and created a folder called libs with libtofi_compute.so and libtofi_config.so copied from "~/Analog\ Devices/TOF_Evaluation_Ubuntu_ADTF3175D-Rel4.3.0/bin_22.04/"
I followed the building steps to create aditofpython.cpython-310-x86_64-linux-gnu.so file and copied it to the site-packages.

SeptimiuVana, 2024-07-16:
Hello, if you used the sd card image from the OF_Evaluation_Ubuntu_ADTF3175D-Rel4.3.0 folder you need to use the 4.3 version on your pc. 

Also you need to provide the ip to the python script. It should be 10.42.0.1 by default if you use network over usb.

Can you try with the other config file?
config_adsd3500_adsd3100.json

Regards,
Septimiu.

kmisiarz, 2024-07-16:
Hi, thanks for the response, indeed the firmware version is TOF_Evaluation_ADTF3175-Rel2.1.1.
So when using 2.1.1, I'm getting the same error on GUI and first_frame.py: The camera was found at /dev/video0, however the response cannot be read. "Failed to read the length of the response string"

The camera is connected with the USB cable (type C to type C).

kmisiarz, 2024-07-17:
[image]

SeptimiuVana, 2024-07-18:
Hello,

Can you try with the latest version? you can download it from here:
https://github.com/analogdevicesinc/ToF/releases

kmisiarz, 2024-08-06:
Hi, we need to use this version because we are using the raw images (part of our study is to develop a different method of combining raw data into 3d images) and as far as I know it's not possible to receive them using the newest firmware. 
1. Do you know if there is any way for me to use the old firmware with Ubuntu 22.04, or
2. Do you know if I can receive the raw data using the newest firmware?