analogdevicesinc/ToF #715: Problem with the aditofpython library

ishopovs, 2024-07-17:
I followed the steps for manually building the SDK on Windows with examples and Python bindings, and the provided C++ examples work. I am testing the installation on Windows 10 with Python 3.12.4, using Visual Studio 2022, CMake 3.30.0, and pybind11 for Python bindings. The version of my SDK is 2.1.1 because it is compatible with the ToF module we have (ADTF3175 Crosby).

However, when I want to test the Python examples, the aditofpython library is not recognized. The setup.py script does not seem to install the aditofpython library correctly. Could you please provide some guidance on how to install aditofpython?

SeptimiuVana, 2024-07-18:
Hello,

That script is for linux only. You should be able to run the scripts from the build folder.
The path should be build/bindings/python/examples.

All the binaries needed should be automatically copied right next to the script that you want to run.

Regards,
Septimiu.

ishopovs, 2024-07-19:
Hi,
Thank you for the explanation. After building the project in Visual Studio, I still do not find the binaries copied in the directories of the Python example scripts. In build/bindings/Release there are three files: aditofpython.exp, aditofpython.lib and aditofPythonNone. I tried copying them next to the scripts but it didn't help. Do you have any suggestions?

SeptimiuVana, 2024-07-22:
Hello, 

Please check if the python bindings are enabled in your cmake files. 
You can set this parameter to on and rebuild the project if not:
https://github.com/analogdevicesinc/ToF/blob/main/CMakeLists.txt#L23

Regards, 
Septimiu.