analogdevicesinc/ToF #746: rawparser.py crash

jkablan, 2024-10-25:
Hi, 

rawparser.py is crashing the python interpreter at line 169.

```point_cloud.points = o3d.utility.Vector3dVector(xyz_frame)``` 

I collected the raw data with the following command and the EVAL-ADTF3175D-NXZ.

``` data_collect.exe --f data_output --m 1 --n 50 --ip 10.43.0.1 config_adsd3500_adsd3100.json ```

I'm running in the aditofpython_env on windows 11. 

Thanks!

NKarnat, 2024-10-28:
Hi @jkablan, 

If you run rawparser.py inside the virtual environment, it might crash. Please try running it outside the virtual environment. Before running, please make sure you install all the dependent modules.

Please try with the latest rawparser.py at [link](https://github.com/analogdevicesinc/ToF/blob/main/tools/misc/rawparser.py)

Regards.

jkablan, 2024-10-28:
Appears to be related to this  https://github.com/isl-org/Open3D/issues/7026 . The issues was resolved by downgrading numpy to version numpy==1.23.5 as suggested in the link.

closed 2024-10-28 by jkablan