analogdevicesinc/ToF #766: Changing the IP address

bndcyan10, 2025-02-07:
Hello,

I have three EVAL-ADTF3175s.
I want to use them with one host PC, so I want to change the IP address of the EVAL-ADTF3175 when connected via USB from 10.43.0.1. 
How can I do this?

My environment is as follows
・SDK v5.0.0
・microSD 5.0.0
・firmware version 5.2.5.

Thank you.

SeptimiuVana, 2025-02-07:
Hello, 

Login to the nxp platform via ssh on that ip address: 10.43.0.1
Open the following file with sudo:
/etc/systemd/network/20-wired-usb0.network

Change the ip from this line to the new ip that you want to use:
https://github.com/analogdevicesinc/ToF/blob/v6.0.0/sdcard-images-utils/nxp/patches/ubuntu_overlay/step3/etc/systemd/network/20-wired-usb0.network#L6

After a reboot the new ip should be set,

Regards,
Septimiu.

bndside, 2025-02-09:
I tried it and it worked, thank you.

bndcyan10, 2025-02-09:
I tried it and it worked, thank you.

closed 2025-02-09 by bndcyan10